ethz-asl/kalibr #18: Building with catkin tools fails

NikolausDemmel, 2015-01-27:
Building with catkin tools (https://github.com/catkin/catkin_tools) instead of `catkin_make` gives "missing include" errors.

```
[aslam_cameras] ==> '/home/den2pal/work/uvm/ros_ws/build/aslam_cameras/build_env.sh /usr/bin/make -j8 -l8' in '/home/den2pal/work/uvm/ros_ws/build/aslam_cameras'
[ 10%] [ 20%] [ 20%] [ 20%] [ 40%] [ 40%] [ 40%] [ 40%] Building CXX object CMakeFiles/aslam_cameras.dir/src/GridCalibrationTargetCheckerboard.cpp.o
Building CXX object CMakeFiles/aslam_cameras.dir/src/GridCalibrationTargetCirclegrid.cpp.o
Building CXX object CMakeFiles/aslam_cameras.dir/src/GridCalibrationTargetObservation.cpp.o
Building CXX object CMakeFiles/aslam_cameras.dir/src/CameraGeometryBase.cpp.o
Building CXX object CMakeFiles/aslam_cameras.dir/src/KeypointBase.cpp.o
Building CXX object CMakeFiles/aslam_cameras.dir/src/GridDetector.cpp.o
Building CXX object CMakeFiles/aslam_cameras.dir/src/FrameBase.cpp.o
Building CXX object CMakeFiles/aslam_cameras.dir/src/Triangulation.cpp.o
In file included from /home/den2pal/work/uvm/ros_ws/src/Kalibr/aslam_cv/aslam_cameras/include/aslam/FrameBase.hpp:6:0,
                 from /home/den2pal/work/uvm/ros_ws/src/Kalibr/aslam_cv/aslam_cameras/src/FrameBase.cpp:1:
/home/den2pal/work/uvm/ros_ws/src/Kalibr/aslam_cv/aslam_cameras/include/aslam/KeypointBase.hpp:10:45: fatal error: sm/kinematics/UncertainVector.hpp: No such file or directory
compilation terminated.
In file included from /home/den2pal/work/uvm/ros_ws/src/Kalibr/aslam_cv/aslam_cameras/src/KeypointBase.cpp:1:0:
/home/den2pal/work/uvm/ros_ws/src/Kalibr/aslam_cv/aslam_cameras/include/aslam/KeypointBase.hpp:10:45: fatal error: sm/kinematics/UncertainVector.hpp: No such file or directory
compilation terminated.
In file included from /home/den2pal/work/uvm/ros_ws/src/Kalibr/aslam_cv/aslam_cameras/src/GridCalibrationTargetObservation.cpp:3:0:
/home/den2pal/work/uvm/ros_ws/src/Kalibr/aslam_cv/aslam_cameras/include/aslam/cameras/GridCalibrationTargetObservation.hpp:12:44: fatal error: sm/kinematics/Transformation.hpp: No such file or directory
compilation terminated.
In file included from /home/den2pal/work/uvm/ros_ws/src/Kalibr/aslam_cv/aslam_cameras/include/aslam/cameras.hpp:5:0,
                 from /home/den2pal/work/uvm/ros_ws/src/Kalibr/aslam_cv/aslam_cameras/src/CameraGeometryBase.cpp:2:
/home/den2pal/work/uvm/ros_ws/src/Kalibr/aslam_cv/aslam_cameras/include/aslam/cameras/CameraGeometry.hpp:6:55: fatal error: sm/kinematics/UncertainHomogeneousPoint.hpp: No such file or directory
compilation terminated.
In file included from /home/den2pal/work/uvm/ros_ws/src/Kalibr/aslam_cv/aslam_cameras/include/aslam/cameras/GridDetector.hpp:12:0,
                 from /home/den2pal/work/uvm/ros_ws/src/Kalibr/aslam_cv/aslam_cameras/src/GridDetector.cpp:9:
/home/den2pal/work/uvm/ros_ws/src/Kalibr/aslam_cv/aslam_cameras/include/aslam/cameras/GridCalibrationTargetObservation.hpp:12:44: fatal error: sm/kinematics/Transformation.hpp: No such file or directory
compilation terminated.
[ 45%] Building CXX object CMakeFiles/aslam_cameras.dir/src/DescriptorBase.cpp.o
make[2]: *** [CMakeFiles/aslam_cameras.dir/src/GridCalibrationTargetObservation.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/aslam_cameras.dir/src/KeypointBase.cpp.o] Error 1
make[2]: *** [CMakeFiles/aslam_cameras.dir/src/FrameBase.cpp.o] Error 1
make[2]: *** [CMakeFiles/aslam_cameras.dir/src/CameraGeometryBase.cpp.o] Error 1
make[2]: *** [CMakeFiles/aslam_cameras.dir/src/GridDetector.cpp.o] Error 1
make[1]: *** [CMakeFiles/aslam_cameras.dir/all] Error 2
make: *** [all] Error 2
[aslam_cameras] <== '/home/den2pal/work/uvm/ros_ws/build/aslam_cameras/build_env.sh /usr/bin/make -j8 -l8' failed with return code '2'
```

And then later also for other packages such as `aslam_backend`... Something with the way `catkin_simple` works, but I am a bit at loss what the correct fix is.

A workaround e.g. for the first error in `aslam_cameras` is adding a manual `include_directories(${catkin_INCLUDE_DIRS}`, but I was under the impression that catkin simple is supposed to do that under-the-hood.

NikolausDemmel, 2015-03-30:
Update: With changes along the lines of #8 and #19 and currently use `catkin_tools` to compile successfully. I have to double check on my other machine if there are any additional tweaks to get that to work.

jbohren, 2015-04-02:
Works for me with `catkin_tools`, too.

NikolausDemmel, 2015-04-03:
Thanks for the report @jbohren. I'll close this then.

closed 2015-04-03 by NikolausDemmel