ethz-asl/kalibr #26: Mistake in computing the Gyro error terms?!

AliAlawieh, 2015-06-15:
Hello,

Once again, I am not sure, but comparing AccelerometerError.cpp and GyroscopeError.cpp under kalibr/aslam_offline_calibration/kalibr/src, why do you pre-multiply by C_b_w in the first file, while you don't in the 2nd file?

omaris, 2015-06-15:
Could you please point us to the respective parts in the code where the
error is evaluated?

AliAlawieh, 2015-06-15:
Sure, it is line 16 of AccelerometerError.cpp:
_predictedMeasurement = C_b_w * (acceleration_w - g_w) + bias;

and line 14 of GyroscopeError.cpp
_predictedMeasurement = angularVelocity + bias;

simonlynen, 2015-06-15:
As you see from the frames of reference of the quantities involved you see that the accel and gravity quantities are expr in the world frame while the bias for accel is in the body frame. Thus you need to transform the quantities into the same frame of ref by premultiplying the rotation matrix C_b_w.

For the gyro this is not needed since the angVel is already given in the body frame.

AliAlawieh, 2015-06-15:
Oh, I thought the angular velocity was just like the acceleration, i.e. expressed in the world frame.
Checking what is passed to the gyroscope error function when called, I found:
wk = poseSplineDv.angularVelocityBodyFrame(tk)
while what is passed to the accelerometer function is:
ak = poseSplineDv.linearAcceleration(tk)
So, you're right... acceleration_w is in the world frame while angularVelocity is in the body frame
Thanks for clearing the confusion :)

closed 2015-06-15 by AliAlawieh