ethz-asl/kalibr #27: should use same pattern size for findChessboardCorners() and drawChessboardCorners()

vanurag, 2015-06-25:
[findChessboardCorners](https://github.com/ethz-asl/kalibr/blob/master/aslam_cv/aslam_cameras/src/GridCalibrationTargetCheckerboard.cpp#L92) uses cv::Size patternSize(cols(), rows());

but [drawChessboardCorners](https://github.com/ethz-asl/kalibr/blob/master/aslam_cv/aslam_cameras/src/GridCalibrationTargetCheckerboard.cpp#L106) uses cv::Size(rows(), cols())

This causes wrong drawing of chessboard corners with different colors in same line like this:
[image]

instead of this:
[image]

omaris, 2015-06-25:
Can you fix this in a pr?

vanurag, 2015-06-25:
sure!

closed 2015-06-25 by vanurag