ethz-asl/kalibr #29: Apriltag family

mhkabir, 2015-08-10:
Hi,
What is the AprilTag family used in the default PDFs available for download?
- Kabir

versatran01, 2015-08-11:
36h11 with a black border of 2

mhkabir, 2015-08-12:
@versatran01 It's weird that I cannot get any of the standard apriltag implementations to detect the board properly.
I tried your "apriltag" package which  I cleaned up and modernized, but tag extraction is not reliable at all for the grids from Kalibr. The tags only get detected at some angles only and are lost soon.
Other 36h11 tags work fine.

versatran01, 2015-08-12:
Yeah, that won't work because the black border is hard coded to 1. Try changing it to 2.

mhkabir, 2015-08-12:
@versatran01 Awesome. Got it. Thanks!
BTW, would you be interested if I cleaned up and maintained/released the TagMapper and associated modules? Already did quite a bit of work on it in order to get it easy to use. Plus documentation as well.

closed 2015-08-12 by mhkabir

versatran01, 2015-08-12:
@mhkabir 
That would be great. Appreciate your work.

mhkabir, 2015-08-18:
@versatran01 
https://github.com/ProjectArtemis/aprilslam :) Done. 
With your permission/confirmation, I'd like to update the licensing on all the files so that other system integrators can work with it. This is impossible without a proper license. What would you suggest/recommend. I was looking at BSD, but happy with the more permissive GPLv3 as well.
Thanks!!

versatran01, 2015-08-18:
@mhkabir 
To be honest, I'm not really familiar with all the licensing stuff, so you can choose whichever one that you see fit.  :smiley:

Zchappie, 2019-10-17:
@mhkabir may I ask what AprilTag implementation did you use? I am currently check the opencv Aruco DICT_APRILTAG_36h11, but it can detect nothing. There may be two reasons for this, one the ASL's tag doesn't have a complete white border around the tag; the other should be the border bits problem. Do you have any idea?

rncy, 2021-01-13:
Actually I wonder why use apriltag with border of 2 rather than "standard" apriltag. For better performance?

SebastianPinzello, 2023-03-24:
@ysklab It might have to do with improved or more robust corner subpixel refinement, but I'm not 100% sure.