ethz-asl/kalibr #31: Failed to obtain orientation prior

art-mx, 2015-10-11:
Hi,
I'm trying to run the calibration of the pixhawk and ueye camera setup with the cde version, but run into the following error: ```Failed to obtain orientation prior!```

The example dataset runs fine though. Any guidance on where to look for the problem would be very appreciated!

```python
./kalibr_calibrate_imu_camera --bag px4-dataset/imu-ueye.bag --cam px4-dataset/camchain.yaml 
--imu px4-dataset/px4_imu.yaml --target px4-dataset/april_6x6_A3.yaml
importing libraries
Initializing IMUs:
  Update rate: 207.0
  Accelerometer:
    Noise density: 0.013 
    Noise density (discrete): 0.187037429409 
    Random walk: 0.000108
  Gyroscope:
    Noise density: 0.00065
    Noise density (discrete): 0.00935187147046 
    Random walk: 2.12e-06
Initializing imu rosbag dataset reader:
	Dataset:          /home/artyom/kalibr-cde/px4-dataset/imu-ueye.bag
	Topic:            /mavros/imu/data
	Number of messages: 18914
Reading IMU data (/mavros/imu/data)
  Read 18914 imu readings over 90.5 seconds                          
Initializing calibration target:
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.035 [m]
    Spacing 0.01225 [m]
Initializing camera chain:
Camera chain - cam0:
  Camera model: pinhole
  Focal length: [416.754436, 416.592321]
  Principal point: [376.0, 240.0]
  Distortion model: equidistant
  Distortion coefficients: [-0.273119, 0.061682, -0.000351, -0.00108]
  baseline: no data available
Initializing camera rosbag dataset reader:
	Dataset:          /home/artyom/kalibr-cde/px4-dataset/imu-ueye.bag
	Topic:            /camera/image_raw
	Number of images: 631
Extracting calibration target corners
  Extracted corners for 631 images (of 631 images)                              

Building the problem
	Spline order: 6
	Pose knots per second: 70
	Do pose motion regularization: False
		xddot translation variance: 1000000.000000
		xddot rotation variance: 100000.000000
	Bias knots per second: 50
	Do bias motion regularization: True
	Blake-Zisserman on reprojection errors -1
	Acceleration Huber width (m/s^2): -1.000000
	Gyroscope Huber width (rad/s): -1.000000
	Do time calibration: False
	Max iterations: 30
	Time offset padding: 0.020000

Estimating imu-camera rotation prior

Initializing a pose spline with 21011 knots (100.000000 knots per second over 210.105000 seconds)
[FATAL] [1444575368.082328]: Failed to obtain orientation prior!
```

rehderj, 2015-10-12:
Hi,

the orientation prior is computed by fitting an orientation curve to measurements provided by the camera and the angular rates measured by the gyroscopes. If the intrinsic calibration of the camera is reasonably accurate and there are no major scaling errors or large temporal offsets in the gyroscope measurements, this works reliably.
So these are all points to starts from: Please provide your report for the camera intrinsic calibration and please ensure that you do not have a scaling error on the rate measurements and that the timestamps are not vastly off. Of these issues, problems with timing are a good candidate to start with.

Cheers, Joern

rehderj, 2015-10-22:
Any luck solving the problem? Can this issue be closed?

art-mx, 2015-10-22:
Haven't solved it yet. I'm suspicious about a few things. 

1) Somewhat asymmetrical camera calibration result, with straight lines on the right side of rectified image not being straight.
[image]

Two different camera calibration results, using ueye and ueye_cam packages.
```javascript
height: 480
width: 752
distortion_model: plumb_bob
D: [-0.2935960992080, 0.08411669991252001, 0.00039760017211, -0.000271737787582, 0.0]
K: [420.5753013752, 0.0, 287.2677867092, 0.0, 420.0058738995, 239.086996746, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [336.567108154, 0.0, 272.100983125, 0.0, 0.0, 373.8412780, 238.612772, 0.0, 0.0, 0.0, 1.0, 0.0]

height: 480
width: 752
distortion_model: plumb_bob
D: [-0.27049, 0.058260000000000006, -2e-05, 8e-05, 0.0]
K: [418.78201, 0.0, 346.79708, 0.0, 418.75386, 241.70869, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [303.31152, 0.0, 329.04409, 0.0, 0.0, 375.34326, 241.62175, 0.0, 0.0, 0.0, 1.0, 0.0]
```

2)A very large ros system timestamp. When running mavros for example, this gives the TF_OLD_DATA warning. No idea how to fix it yet.

rehderj, 2015-10-23:
Ok. I presume that these are two results for the same camera, in which case the difference in the estimation of the principle point is significantly too large. It further looks like that you obtained this calibration with the ros utility rather than with kalibr.
Possible problems are that you did not sufficiently sample the "checkerboard pose space" while calibrating. Personally, I collect multiple minutes of data while presenting the calibration target in various orientations and at different distances. A more principled approach can be found in  "AprilCal: Assisted and repeatable camera calibration" by Richardson et al. . While they guide the user, kalibr takes a different approach by selecting views from the pool of data that contribute most to observing the calibration parameters. I think that the equidistant distortion is a better model of the effects present in most optics, a claim that seems to also be supported by the aforementioned publication.
Please record a sufficient amount of data and try the kalibr calibration function using the equidistant model to see whether it fixes your issues. I noticed that for this model, the distortion parameters vary widely for similar lenses while giving similar, low reprojection errors. Accordingly, please assess the quality of your calibration by the reprojection error on a test ste and the reproducibility of the focal length and principle point rather than the numerical values of the distortion parameters.
 
Regarding 2), this could quite possibly be the source of your problems, but it is not related to this software package. Please reconsider whether the problem that you are observing are rooted in our calibration toolbox or in your current system architecture.

Cheers!

art-mx, 2015-10-27:
Hi, camera calibration result seems ok now, after running it with the kalibr tool and checking it with calibration validator. The cam-imu calibration also worked with a larger marker board. Thank you for support!

closed 2015-10-27 by art-mx