ethz-asl/kalibr #41: Optimization failed in kalibr_calibration_imu_camera

francescoli, 2016-02-04:
I've tested 5 different dataset acquired today, all failed, but not the sample dataset nor the ones I tested yesterday. Is it caused by the dataset and how to avoid it?
```
The Jacobian matrix is 194404 x 25179
[0.0]: J: 1.45862e+07
[1]: J: 2.21318e+06, dJ: 1.2373e+07, deltaX: 0.991163, LM - lambda:100 mu:2
[2]: J: 33328.9, dJ: 2.17985e+06, deltaX: 0.521266, LM - lambda:33.3333 mu:2
[3]: J: 28162.1, dJ: 5166.75, deltaX: 0.0966487, LM - lambda:11.1111 mu:2
[4]: J: 13295.2, dJ: 14866.9, deltaX: 0.0136785, LM - lambda:12.7412 mu:2
Exception in thread block: Exception in thread block: [aslam::Exception] /home/louis/catkin_ws/src/kalibr/aslam_nonparametric_estimation/aslam_splines/src/BSplineExpressions.cpp:447: toTransformationMatrixImplementation() assert(_bufferTmin <= _time.toScalar() < _bufferTmax) failed [15106.3 <= 15106.3 < 15106.4]: Spline Coefficient Buffer Exceeded. Set larger buffer margins!
[aslam::Exception] /home/louis/catkin_ws/src/kalibr/aslam_nonparametric_estimation/aslam_splines/src/BSplineExpressions.cpp:447: toTransformationMatrixImplementation() assert(_bufferTmin <= _time.toScalar() < _bufferTmax) failed [15118.2 <= 15118.2 < 15118.2]: Spline Coefficient Buffer Exceeded. Set larger buffer margins!
Exception in thread block: [aslam::Exception] /home/louis/catkin_ws/src/kalibr/aslam_nonparametric_estimation/aslam_splines/src/BSplineExpressions.cpp:447: toTransformationMatrixImplementation() assert(_bufferTmin <= _time.toScalar() < _bufferTmax) failed [15094.8 <= 15094.8 < 15094.9]: Spline Coefficient Buffer Exceeded. Set larger buffer margins!
[ERROR] [1454555378.858709]: Optimization failed!
Traceback (most recent call last):
  File "/home/louis/catkin_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 206, in <module>
    main()
  File "/home/louis/catkin_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 182, in main
    iCal.optimize(maxIterations=parsed.max_iter, recoverCov=parsed.recover_cov)
  File "/home/louis/catkin_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py", line 183, in optimize
    raise RuntimeError("Optimization failed!")
RuntimeError: Optimization failed!
```

rehderj, 2016-02-04:
Please increase the timeOffsetPadding (https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera#L171) and see whether this fixes your issue. In general, this parameter is the margin within the temporal offset is allowed to vary.

francescoli, 2016-02-04:
Yes, it fixes. Is there any rule of setting this parameter, or just increase it when the same issue occurs.

rehderj, 2016-02-04:
It is a trade-off between computation time and robustness: If you set the margins conservatively large, it will result in increased computation time, if you set it too low, the estimate might travel across knot boundaries, violating the precomputed sparsity pattern. Please see the implementation section in Paul's paper for details on that (https://3234f89137bccf2ede29cc86e315c75116020d70.googledrive.com/host/0B64GJ60h3Ai1MVVwWTZwekhtcFU/publications/bib/furgale_iros13.pdf).
The "trajectory" of your estimate of the temporal offset during optimization will depend on the shape of the cost surface which in turn is related to the very motion you performed. Hence, it is difficult to provide a rule of thumb here.

francescoli, 2016-02-04:
Got it, thank you Joern

closed 2016-02-04 by francescoli

andre-nguyen, 2016-03-02:
@francescoli Out of curiosity (actually facing this problem myself) what value did you increase the temporal offset to?

rehderj, 2016-03-02:
While this parameter might need some adaptation, repeated problems setting the margins correctly often hint to issues in the data, for example due to spurious measurements, incorrect scaling or incorrect timestamping.
Please consider running the calibration with the ``--verbose`` option (https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera#L92), which will plot the cross correlation used for finding an initial estimate (https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py#L217-L229). If the plot does not exhibit a distinct peak, please specifically check your gyroscope data, since this is being used here.

gethubwy, 2019-07-10:
Hello,rehderj , I also met this problem, I want to know  the value  timeOffsetPadding should be  what value?
maybe 100?1000?  or whatever?

shakeelsamsu, 2019-07-15:
@gethubwy I've been using a value of 0.1 (100 ms), as per this [issue](https://github.com/ethz-asl/kalibr/issues/228#issuecomment-425361720). Note, however, that this is simply a buffer margin for the timestamps. Oftentimes, it might be helpful to review your dataset and ensure that there are no problems there instead of changing this value.

suho0515, 2020-06-16:
> Please increase the timeOffsetPadding (https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera#L171) and see whether this fixes your issue. In general, this parameter is the margin within the temporal offset is allowed to vary.

thanks, solved it with changing to 100ms

egemenertugrul, 2022-12-15:
PS: It is possible to use the ``--timeoffset-padding`` argument when running ``kalibr_calibrate_imu_camera``.