ethz-asl/kalibr #48: omnidirectional camera model conversion

ChristophTobler, 2016-05-18:
Not really an issue but:
Is there a way to convert the intrinsics and distortion coefficients to pinhole model after calibration?

The problem is this: I need the omni model for the calibration to work with 8 (4 stereo in all directions) cameras (small overlap over a 90 deg corner). But afterwards I want to triangulate in the centre of the images (which is more or less pinhole) and for that I need the pinhole model. I want to avoid calibrating it twice (first all 8 with omni and then stereo with pinhole).

I had a look at the paper "A Generic Camera Model and Calibration Method for Conventional, Wide-Angle, and Fish-Eye Lenses" but could find anything helpful...

ZacharyTaylor, 2016-05-19:
Converting between the two representations is a non-trivial problem as the two methods produce curves that cannot be mapped into the other representation exactly.

However, it would be possible for a specified range to find the representation that gives the smallest difference in distortion when transferring from one representation to the other.

I have a script I wrote in matlab that might be a bit easier for you to follow then looking at the kalibr code if you want to understand a bit more about how the distortion is calculated. I also had a quick 5 min go at writing a script to convert from equidistant to radtan using a least squares metric. I can't say how accurate this approach would be but it might help you to do this conversion.
[Undistort.m.txt](https://github.com/ethz-asl/kalibr/files/272054/Undistort.m.txt)
[Equi2Pin.m.txt](https://github.com/ethz-asl/kalibr/files/272077/Equi2Pin.m.txt)

ChristophTobler, 2016-05-19:
I'll have a look at it. Thanks!

closed 2016-08-26 by rehderj