ethz-asl/kalibr #50: Multiple IMU calibration configuration unclarity

timonh, 2016-06-17:
Hi all,

I am looking into the calibration of two imu's using KALIBR to obtain an estimation of their relative spacial transform. (Imu1 with respect to imu0)

My question is: For performing the multi imu calibration: do I move the robot and bag the two IMU-topics (imu0 and imu1) and run the calibration or do I move the robot and bag the two IMU-topics as well as two camera topics (with an aprilgrid in the field of view) thus -> imu0, imu1, cam0, cam1.

I think this might need more clarification in the Wiki.

Thanks!

T

rehderj, 2016-06-17:
Hi,

thank you for your interest in our work. While some documentation is still missing, I believed this to be contained in our wiki. Please see https://github.com/ethz-asl/kalibr/wiki/Multi-IMU-and-IMU-intrinsic-calibration#2-dataset-collection which in turn links to https://github.com/ethz-asl/kalibr/wiki/Camera-IMU-calibration#2-collect-images which states that you should indeed record at least one image stream with your IMUs.
The multi-IMU calibration is a straight-forward *extension* of kalibr which requires the exact same work flow with the difference that you may record and calibrate multiple IMUs, IMU intrinsics and size effect.

Cheers, Joern

rehderj, 2016-06-17:
@timonh Can this issue be closed?

closed 2016-06-19 by rehderj