ethz-asl/kalibr #51: Calibration target on A4 paper

shanmo, 2016-07-12:
Hi,

I would like to print the calibration target, such as an Aprilgrid or a checkerboard, on A4 paper and use it for calibration. Could anyone kindly advice how to use kalibr_create_target_pdf to specify the size? 

After we have the calibration target, how could we set the size in the yaml accordingly? Should we use the size values in the kalibr_create_target_pdf directly or measure it on the actual calibration target printed on the A4 paper?

Any help is appreciated!

rehderj, 2016-07-12:
Hi,
using the `--help` option, you will be presented with a set of options to specify the target, including size.
The script should yield a pdf with the specified dimensions. You should then choose options for printing that avoid scaling the document during printout.

Please set the **real (measured) dimensions** in the yaml. Otherwise you will end up with scaled estimates of your baseline and your camera/IMU calibration will likely fail, since it will be impossible to reconcile  accelerations perceived in metric units with the scaled estimates of distances.

Please consider closing the issue if this answered your question.
Cheers!

shanmo, 2016-07-12:
Thanks for your reply! 

In fact, I am not quite sure what sizes should be used for A4 paper. Could I use the example Calibration targets provided on the [wiki](https://github.com/ethz-asl/kalibr/wiki/downloads), such as the Aprilgrid 6x6 0.8x0.8 m (A0 page), print it out on A4 paper, and then measure the sizes of the target on the A4 paper for the yaml?

rehderj, 2016-07-12:
In general, this would be a valid approach. However, you might experience difficulties identifying such a small target at larger distances depending on the resolution of your camera. Furthermore, the estimation problem should in general be better constrained the more of the area in the image is covered by the target.

It is almost always a good idea to print the target as large as feasible.

shanmo, 2016-07-13:
@rehderj I see, thanks so much for your advice!

closed 2016-07-13 by shanmo