ethz-asl/kalibr #52: Crash with Fov distortion model

fejj, 2016-07-13:
Hi,

The kalibr_calibrate_cameras command seems to fail when called with the fov distortion. I have been able to repeat the error message below on 3 different computers, with Kalibr installed from 3 different people. This issue doesn't appear if the --show-extraction option is enabled or with the other distortion models.

`kalibr_calibrate_cameras --bag ~/Desktop/2016-07-12-23-25-53.bag --topics /camera/image_raw --models pinhole-fov --target ~/april_6x6.yaml 
importing libraries
Initializing cam0:
	Camera model:	  pinhole-fov
	Dataset:          /home/cbrommer/Desktop/2016-07-12-23-25-53.bag
	Topic:            /camera/image_raw
	Number of images: 671
Extracting calibration target corners
python: /usr/include/boost/archive/detail/oserializer.hpp:436: static void boost::archive::detail::save_pointer_type<Archive>::polymorphic::save(Archive&, T&) [with T = aslam::cameras::CameraGeometryBase; Archive = boost::archive::text_oarchive]: Assertion `__null != bpos' failed.
Aborted (core dumped)`

rehderj, 2016-07-15:
Could you please check whether https://github.com/ethz-asl/kalibr/pull/53 addresses your issue?
Thanks!

fejj, 2016-07-15:
Yes, it did solve it. Thanks for the fast fix!

closed 2016-07-15 by fejj