ethz-asl/kalibr #55: Optimization failed!

Narcissuscyn, 2016-07-18:
Can anyone tell me how to solve this problem:

Exception in thread block: [aslam::Exception] /var/kalibr-build/src/kalibr/aslam_nonparametric_estimation/aslam_splines/src/BSplineExpressions.cpp:447: toTransformationMatrixImplementation() assert(_bufferTmin <= _time.toScalar() < _bufferTmax) failed [763.696 <= 763.746 < 763.746]: Spline Coefficient Buffer Exceeded. Set larger buffer margins!
Exception in thread block: [aslam::Exception] /var/kalibr-build/src/kalibr/aslam_nonparametric_estimation/aslam_splines/src/BSplineExpressions.cpp:447: toTransformationMatrixImplementation() assert(_bufferTmin <= _time.toScalar() < _bufferTmax) failed [748.448 <= 748.499 < 748.498]: Spline Coefficient Buffer Exceeded. Set larger buffer margins!
Exception in thread block: [aslam::Exception] /var/kalibr-build/src/kalibr/aslam_nonparametric_estimation/aslam_splines/src/BSplineExpressions.cpp:447: toTransformationMatrixImplementation() assert(_bufferTmin <= _time.toScalar() < _bufferTmax) failed [756.737 <= 756.787 < 756.787]: Spline Coefficient Buffer Exceeded. Set larger buffer margins!
[ERROR] [1468850859.791703]: Optimization failed!

thanks very much!

rehderj, 2016-07-18:
Hi,
thank you for using our software. Please see other issues (also closed ones and in particular https://github.com/ethz-asl/kalibr/issues/41) for explanations on that issue.
Also, in the "extended-kalibr" branch,  there is an option of adapting the parameter: https://github.com/ethz-asl/kalibr/wiki/Multi-IMU-and-IMU-intrinsic-calibration#3-running-the-calibration .
**However, that you are experiencing this problem likely hints to an underlying issue with your dataset.**
Cheers!

Narcissuscyn, 2016-07-18:
Hi,
     Thanks a lot!
     I have tried many datasets,sometimes it  failed,or got unreasonable results.I am not sure whither the imu.yaml is correct.I get noise desity  from the datasheet and the randwalk from allan deviation plot.I increased the timeOffsetPadding,it woks with the full dataset,but  it failed with fewer data with --bag-from-to .

rehderj, 2016-07-18:
The values set in imu.yaml will have an impact on the precision, but in our experience, they can set within a broad range without leading to failures in the calibration. Hence, it is more likely that there are issues with your dataset.
As always, I would suspect **timing** as the culprit. We have posted extensively on this in the issues and also addressed it in a recent work (http://ieeexplore.ieee.org/xpl/articleDetails.jsp?arnumber=7445934) so please see these for starting point.

Narcissuscyn, 2016-07-18:
Issuses with my dataset? can you  give me some examples? I have strictly organized the input datasets as  Requirements and  can't find any problem. Always thanks for your advice，i will refer to this material.

Narcissuscyn, 2016-07-19:
thanks for your advice! I managed to solve the problem by  adjust the value of noise density and bias random walk. The result is good now.

closed 2016-07-19 by Narcissuscyn

Yvala, 2016-07-19:
Thanks for this conversation. It was a great help for me as well. However, just from curiosity, what can be the issues with my dataset? Some examples would be helpful, since I think I have my noise density and bias random walk set properly.

rehderj, 2016-07-19:
Please see for example issue https://github.com/ethz-asl/kalibr/issues/46 for suggestions. In our experience, noise parameters and the specific motion play a less important role (within bounds of course). Commonly, inadequate system design is the main cause for bad results.

Here an incomplete list of possible problems:
-  incorrect timestamping fo the measurements
- use of rolling shutter cameras (not supported right now)
- poorly calibrated camera intrinsics
- spurious measurement from the IMU
- motion blur

Please check your data for any of these issues.

Cheers!