ethz-asl/kalibr #57: Unrealistic output

Yvala, 2016-07-25:
Hello,

I am using camera calibration and further camera-imu calibration on 3 different smartphones. However, only one gives me the right position of imu vs. camera. The other two say the IMU is more than 15cm away from camera in x and y axes and more than 3 cm in z axis. I am not sure why is this not working only on the two of them (iphone 5 and nexus 5). Do you have some suggestions?

Thank you

rehderj, 2016-07-25:
Could you please provide more information?
Specifically:
- How do you timestamp your data?
- How did you set the exposure times of the cameras?
- What IMU model did you pick?
- Please provide the result pdfs here.

A list of aspects to consider can be found here: https://github.com/ethz-asl/kalibr/issues/55#issuecomment-233626586
In your case, specifically the rolling shutter issue is of interest, since the vast majority of smartphone cameras are rolling shutter devices.

Yvala, 2016-07-25:
I am very sorry but I have no idea what are the answers. I was given bag files and yaml files for IMUs and I am trying to figure it out. 
1) What do you mean by timestamping the data? Every smartphone has its own timestamp, I do not know the basis of that.
2) What do you mean by setting the exposure times? Do you mean the extrinsics and intrinsics?
3) Is there a possibility to choose a type of IMU? 

I attached the two pdfs for iphone and nexus and the text files with outputs.

I am really sorry to bother with such beginners questions but I am trying to understand the problem.

[nexus-report-imucam.txt](https://github.com/ethz-asl/kalibr/files/381419/nexus-report-imucam.txt)
[iphone-report-imucam-49.txt](https://github.com/ethz-asl/kalibr/files/381420/iphone-report-imucam-49.txt)

[iphone-report-imucam-49.pdf](https://github.com/ethz-asl/kalibr/files/381417/iphone-report-imucam-49.pdf)
[nexus-report-imucam.pdf](https://github.com/ethz-asl/kalibr/files/381418/nexus-report-imucam.pdf)

rehderj, 2016-07-25:
Unfortunately, we cannot provide any meaningful help for problems arising from issues in the design or understanding of the system providing the data input to kalibr.

Please read our wiki and check our publications for deeper insights, specifically on the issue with [varying exposure time](https://www.semanticscholar.org/paper/Unified-temporal-and-spatial-calibration-for-multi-Furgale-Rehder/4466dcf725e5383b10a6c5172c860f397e006b48/pdf), with [rolling shutter cameras](http://www.cv-foundation.org/openaccess/content_cvpr_2013/papers/Oth_Rolling_Shutter_Camera_2013_CVPR_paper.pdf), and with [IMU intrinsics](http://ieeexplore.ieee.org/xpl/articleDetails.jsp?arnumber=7487628).

If it was ok with you, I would suggest closing the issue for now. If you are still experiencing problems despite having addressed the aforementioned aspects, you could just reopen it.

closed 2016-07-25 by Yvala