ethz-asl/kalibr #59: A problem about the matrics in the result

Narcissuscyn, 2016-07-28:
Hi,
    I get a matrics in the result,but it seems unreasonable.Is it right that the last column [0.00448395,-0.00498745,-0.0022667]-1 represents the translation matrics[x,y,z]-1?I think the value is too small,the real distance of IMU and Camera is about [0.07,0.035,0.02] by measurement.but other values are ok! 
T_ci:  (imu to cam0): [m]
[[ 0.97904792  0.13214156 -0.15493151  0.00448395]
 [ 0.17046857 -0.94803377  0.26864929 -0.00498745]
 [-0.11138056 -0.28943148 -0.95069647 -0.0022667 ]
 [ 0.          0.          0.          1.        ]]
     I am looking for your advises! Thanks!

Narcissuscyn, 2016-07-28:
I have an another question here.Should i strictly make the camera at a rate of 20Hz and IMU at a rate of 200Hz?

rehderj, 2016-07-28:
Without any additional context, there is very little specific advice we ca give. Also, from the discrepancy between your results and what you expected, I suspect some more fundamental issues in the design or understanding of the system that you used to acquire the data with. Unfortunately, this is something that we cannot really help with.

So just a couple of high-level points here:
- camera and IMU rate just have to be sufficiently high, they do not have to match the settings used in our example. We have successfully tested different IMU rates between 100 Hz and 2 kHz and camera rates between 7.5 Hz and 30 Hz.
- Personally, I *never* do a calibration only once, but instead use multiple datasets from the same device to get an idea about the distribution of the estimates. Please see our publications for references on what distributions to expect here. If your results are not within this ballpark, there is likely something that you can improve in terms of system design, system understanding or dataset collection.
- Please see other issues (including closed ones) as well as our wiki and the publications referenced there for hints on what is important in terms of system design and dataset collection.

Please consider closing this issue in case you are able to confirm that the problem is rooted in the input data.

Cheers!

Narcissuscyn, 2016-07-29:
Thanks a lot! 
   But i am still not clear about  the last column in the matrics.Is it represents the real distance(dx,dy,dz) between in the IMU and camera?

rehderj, 2016-07-29:
Personally, I cherish any exchange about the toolbox and I would like to encourage you and other users to reach out to us if you experience issues with kalibr or if we missed adding practical advice to our wiki that cannot be extracted from the publications referenced there.

However, I hope that you understand that we lack the resources to provide help on fundamentals. Please read our publications and their related work for clarity on the parameters that are estimated by kalibr.

If it was ok with you, I would like to close this issue now. Please do not hesitate to contact us if specific issues with the toolbox arise.

closed 2016-07-29 by rehderj