ethz-asl/kalibr #63: Obtaining IMU parameterss from datasheet

XinkeAE, 2016-07-28:
Hi,

Thanks for sharing the package with the community! It helps a lot!

Currently we are trying to find the corresponding parameters of our IMU according to:
https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model

According to the datasheet of ADIS 16448 (http://www.analog.com/media/en/technical-documentation/data-sheets/ADIS16448.pdf)
The parameters are:
Gyroscopes:
    Angular Random Walk: 0.66 deg/sqrt(hr)
    Rate Noise Density (f=25Hz): 0.0135 deg/sec/sqrt(hz)
Accelerometer:
    Velocity Random Walk: 0.11 m/sec/sqrt(hr)
    Noise Density: 0.23 mg/sqrt(hz)

And the parameters for the same IMU in the Kalibr example (https://github.com/ethz-asl/kalibr/wiki/yaml-formats) are:
Gyroscopes:
    Random Walk: 4.0e-6 rad/(s^2)/sqrt(hz)
    Noise Density(continuous time): 0.005 rad/s/sqrt(hz)
Accelerometer:
    Random Walk: 0.0002 m/(s^3)/sqrt(hz)
    Noise Density(continuous time): 0.01m/(s^2)/sqrt(hz)

Could you please specify the relationship between the two sets of parameters? This may be add to the tutorial as an example.

Thank you so much!

- Xinke

nikolicj, 2016-07-29:
Dear Xinke

Thank you for your inquiry.

**ADIS16448 DATASHEET**

Most datasheets specify values for what we refer to as the "white noise" or "broadband noise" strength (`gyroscope_noise_density` (sigma_g) and `accelerometer_noise_density` (sigma_a)) in Kalibr's IMU noise model. The datasheet you to ([http://www.analog.com/media/en/technical-documentation/data-sheets/ADIS16448.pdf](http://www.analog.com/media/en/technical-documentation/data-sheets/ADIS16448.pdf)) refers to these values as `Rate Noise Density` for the gyros and as `Noise Density` for the accles (Table I). The units are (up to scale) rad/s/sqrt(Hz) and m/s^2/sqrt(Hz), which matches our model.

Datasheets rarely (never) specify the noise model parameters corresponding to the "bias part" of Kalibr's gyro and accel models. The reason for this it that real gyros and accels do not exhibit noise characteristics which match the standard model we (and many others) use. This is why the Allan deviation is often provided.

The "Angular Random Walk" (gyro) and the "Velocity Random Walk" are **not** related to the "bias part" of Kalibr's IMU model. These values say little about the long-term bias fluctuations. Analog devices specifies them as the Allan deviation at an integration period of 1 s (see https://ez.analog.com/docs/DOC-2163). And you can indeed make to following computation:

Take the "Rate Noise Density", specified as 0.0135deg/s/sqrt(Hz), virtually integrate this idealized "white noise process" over a period of 1 s, and scale to the units of the "Angular Random Walk" provided in the datasheet (deg/sqrt(Hr)):

0.0135 * sqrt(3600) => 0.81 deg/sqrt(Hr), vs. 0.66 deg/sqrt(Hr) in the datasheet.

The small discrepancy might be explained by the fact that in reality the "white noise" is rather some "broadband noise" and does not have a flat power spectrum. You can similarly verify this for the accelerometer.

In conclusion: You can obtain the `gyroscope_noise_density` and the `accelerometer_noise_density` directly from Analog Device's datasheet, but not the `gyroscope_random_walk` and the `accelerometer_random_walk` (except from the Allan variance given in the datasheet, or very indirectly from the "in-run bias stability", if you are willing to make a few additional engineering-type assumptions).

Does that answer your question?

**ADIS16448 PARAMETERS FOR KALIBR**

The parameters specified in https://github.com/ethz-asl/kalibr/wiki/yaml-formats are not used by us anymore. We will replace these with the ones we use for calibrations involving the ADIS16448, and with which @rehderj obtains good results (see [1]). They can be obtained with the method proposed in the wiki. The values for the `gyroscope_noise_density` and the `accelerometer_noise_density` roughly match the values provided in the datasheet, and you can use one or the other.

`gyroscope_noise_density`
proposed here: 1.87 * 10^-4 rad/s/sqrt(Hz)
datasheet: 2.36 * 10^-4 rad/s/sqrt(Hz)

`accelerometer_noise_density`
proposed here: 1.86 * 10^-3 m/s^2/sqrt(Hz)
datasheet: 2.26 * 10^-3 m/s^2/sqrt(Hz).

Let uns know if you need more information on this, Xinke.

[J. Rehder: Extending kalibr: Calibrating the Extrinsics of Multiple IMUs and of Individual Axes, Table II](http://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=7487628)

[image]

XinkeAE, 2016-07-29:
Hi,

Thanks for the quick reply. It was very helpful!

Xinke

closed 2016-07-29 by XinkeAE

CtfChan, 2018-01-30:
Hi what did you mean when you said that you can obtain the random walk parameters "very indirectly from the in-run bias stability"?

cynosure4sure, 2018-03-31:
Can anyone explain the process of obtaining random walk parameters from in-run bias stability? Thank you

mnissov, 2023-04-20:
I think the [original answer](https://github.com/ethz-asl/kalibr/issues/63#issuecomment-236140184) made an error in getting PSD from the datasheet. Though I'm not 100% sure so I welcome corrections.

The first problem was in suggesting that "rate noise density" and "noise density" are the analogues to gyro/accel noise density as kalibr expects. As the former quantities are RMS values in this case. My impression is that the value most resembling the gyro/accel PSD (which is what kalibr wants) is the "angular random walk" and "velocity random walk" in this datasheet. 

I think this error is further reflected in the comparison of datasheet values with empirically estimated values. If we say the values found in [the referenced paper](http://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=7487628) are trustworthy, then you can see the random walk values get much closer to the empirically estimated values:

`gyro PSD`
proposed here: 1.87 * 10^-4 rad/s/sqrt(Hz)
datasheet (`rate_noise_density`): 2.36 * 10^-4 rad/s/sqrt(Hz)
datasheet (`angular_random_walk`): (0.0135 / sqrt(3600) * pi/180) = 1.92 * 10^-4 rad/sqrt(s)

`accelerometer PSD`
proposed here: 1.86 * 10^-3 m/s^2/sqrt(Hz)
datasheet (`noise density`): 2.26 * 10^-3 m/s^2/sqrt(Hz).
datasheet (`velocity_random_walk`): (0.11 / sqrt(3600)) = 1.83 * 10^3 m/s/sqrt(s)

Note also the units match, `rad/sqrt(s)` == `rad/s/sqrt(Hz)` and `m/s/sqrt(s)` == `m/s^2/sqrt(Hz)`

My impression is the terminology is quite often misused when talking about these quantities, but at least here "angular random walk" and "velocity random walk" is consistent with Woodman's "An introduction to inertial navigation". Also, it is consistent with the kalibr [wiki][(https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model#the-noise-model-parameters-in-kalibr)