ethz-asl/kalibr #66: No module named scipy.optimize

v0n0, 2016-08-06:
Hey,

I just pulled down the latest version (added 3 commits) and I recompiled. After that I resourced the workspace and tried to run cam+imu calibration again, but I got this error:
```
Traceback (most recent call last):
  File "/home/xxx/kalibr_workspace/devel/bin/kalibr_calibrate_imu_camera", line 6, in <module>
    exec(fh.read())
  File "<string>", line 5, in <module>
  File "/home/xxx/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/__init__.py", line 4, in <module>
    import IccSensors as sens
  File "/home/xxx/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 16, in <module>
    import scipy.optimize
ImportError: No module named scipy.optimize
```
This is how I solved:
```
sudo apt-get install gfortran
sudo -H pip install scipy
```
These dependencies should be added to the [installation guide](https://github.com/ethz-asl/kalibr/wiki/installation) IMHO.

imseancas, 2016-08-26:
I got the error:"No module named aslam_cv" when I tried to make the camera calibration like this: "rosrun kalibr kalibr_calibrate_cameras --bag calibration.bag --topics /usb_cam/image_raw models pinhole-equi target april.yaml"

rehderj, 2016-08-26:
Thank you. I added `python-scipy` to the dependencies.

@imseancas Your error seems unrelated to this. Please see https://github.com/ethz-asl/kalibr/wiki/multiple-camera-calibration for details on how to run the framework. Did you source the respective workspace?

imseancas, 2016-08-29:
Yes, I compiled the package in my workspace(catkin_ws) and souced it by "source ~/catkin_ws/devel/setup.bash"

v0n0, 2016-08-29:
Going to close this as it was about a different issue. @imseancas please open a new one.

closed 2016-08-29 by v0n0