ethz-asl/kalibr #71: RS calibration failing

v0n0, 2016-08-26:
Hi @rehderj,

thanks for merging in RS and the dataset, I was able to run your 6Hz dataset with latest `master`. 
```
LineDelay:
-6.00622067443e-07
```
With a framerate of 20 it completes successfully, while if I try 6 if fails (Spline Coefficient Buffer Exceeded).

Unfortunately, it doesn't work with any of my datasets, with any framerate. I made a dataset around the same length as yours, same movements, but I just get a `NaN` Jacobian from the optimization at the first step and then it fails.

I wonder if I'm creating the bag in some kind of wrong way: do you use JPEG files or raw images? I pass JPEGs to the bagcreater script. Configuration for the target is as follows:
```
target_type: 'aprilgrid' #gridtype
tagCols: 6               #number of apriltags
tagRows: 6               #number of apriltags
tagSize: 0.088           #size of apriltag, edge to edge [m]
tagSpacing: 0.3          #ratio of space between tags to tagSize
```
I measured the tag manually and it is 88mm indeed.

What else is left to check? How should I calculate the estimated inverse variance of the feature detector?

Thanks a lot

rehderj, 2016-08-26:
Mmh, it is odd that you would get a negative line delay.
I was not fully sure about this statement of yours:
> With a framerate of 20 it completes successfully, while if I try 6 if fails (Spline Coefficient Buffer Exceeded).

Do you mean when setting the script option? Luc's documentation suggests that this parameter provides some control over the knot sequence governing the spline: https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/python/kalibr_rs_camera_calibration/RsCalibrator.py#L70-L73
Accordingly, it might also somewhat depend on the motion and rather than just setting the exact framerate it might be worth varying the parameter a little.

Do you detect the target in every single frame? I am just guessing, but your problems could also arise from missing detections and consequently gaps in the support of the spline. In that case, you could also try the dot grid pattern as recommended in the respective paper.

Finally, calibrating the line delay _without_ an IMU appears to be a much more difficult problem than when an IMU is present (at least to me). The reason for that lies in the fact that you would like to "constrain" the spline in a way that discourages the estimation of infeasible motions (think about it, if infinite linear and angular accelerations were a possibility, the camera could basically teleport itself to poses that minimize the reprojection error).
Luc addressed this concern with regularizer terms and some clever knot placement. However, if you had an IMU, you could directly "constrain" the accelerations. Accordingly, it might still be a valid approach to push ahead with the transfer of RS error terms into the camera/IMU calibration although the camera calibration currently fails for you.

v0n0, 2016-08-26:
Yeah I meant "passing 20 as a framerate to the script".

Your suggestion about the features not always being detected was on point: I recorded a dataset making sure the full target was always visible and it started to run. I will have to do another one with more light, but it seems this was the main blocker. I guess the optimization was indeed trying to teleport ;)

I will try another dataset and then set to do the change to the code for cam + IMU.

Thanks again, very helpful.

closed 2016-08-26 by v0n0