ethz-asl/kalibr #76: Allan Standard Deviation Script

Daniel-Eckert, 2016-10-10:
Hi there,

I am currently trying to use kalibr with the Invensense MPU-9250 IMU. 
As described here (https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model), I want to obtain the random walk parameters from  the Allan Standard Deviation plot.

On that page you mention an AD script in the following ToDo: [TODO(nikolicj) add AD script]
Would it be possible to get access to that script? This would really be amazingly helpful.

Thanks so much for providing this great software!

Cheers,
Daniel

rehderj, 2016-10-11:
Hi Daniel,

thanks for using our tool. Unfortunately, Janosch has left the group, so he will be unable to provide such script. However, there is a host of Matlab implementations that can be found on Mathworks File Exchange. If you prefer a python environment, you may want to try Allan Tools (https://pypi.python.org/pypi/AllanTools/2016.3). Feel free to contribute python based scripts back to the repository. They may be helpful to others. 

Cheers, Joern

goldbattle, 2022-04-22:
There is a wide range of projects that try to help with this.
For example:
- https://github.com/ori-drs/allan_variance_ros
- https://github.com/AlbertoJaenal/imu_still_calibration
- https://github.com/rpng/kalibr_allan
- https://github.com/GAVLab/allan_variance

goldbattle, 2022-05-03:
Wiki has been updated.

closed 2022-05-03 by goldbattle