ethz-asl/kalibr #79: virtual memory exhausted:

zhubaohua, 2016-10-18:
hello,
I am currently trying to use kalibr,I choose "building from source" when installing the toolbox,
  but the project building failed with the error:   virtual memory exhausted.
I find the reason is   " /Kalibr/aslam_cv/aslam_cv_backend_python/src/module.cpp"
when I comment  module.cpp ,building will success. 
but i need build module.cpp,otherwise I can't  use  tools .
[image]

HannesSommer, 2016-10-18:
Unfortunately this particular file currently requires quite some memory to compile. (I reckon at least 4GB)
Maybe you can manage to free enough or add  swap space. Add an  `-j 1` argument to your build command to compile this file. 
If all that fails you could try to split the file in multiple compilation units (cpp) or remove parts that you don't need for your task. It should be possible to delete blocks like https://github.com/ethz-asl/kalibr/blob/master/aslam_cv/aslam_cv_backend_python/src/module.cpp#L54-L56 if you don't need them for your setup.

zhubaohua, 2016-10-19:
thank you very much ,I  have solved this issue when i build this code in another PC, maybe the reason is hardware . Besides，what does  the parameter "cam_overlaps"  in  camchain.yaml of "IMU-camera calibration"  mean? I can't find that in  YAML.formats

rehderj, 2016-10-19:
`cam_overlap` identifies those cameras that share some parts of their field of view. The overlap is identified from your dataset. For this to work, the target must be identified in images from both cameras that have approximately the same timestamp. Accordingly, it might fail if your cameras are not properly synchronized or for small overlap when the target can not be reliably identified in the peripherals of the views.  In general, you will get a more robust estimate of the overlap when using kalibr with the April tag board for which not the entire target has to be in the field of view of the camera.

closed 2022-04-22 by goldbattle