ethz-asl/kalibr #80: Could not read configuration from imu_adis16448.yaml

wyycv, 2016-10-30:
Hello,I have successfully installed kalibr from source and downloaded your sample datasets to run the example using the following command
kalibr_calibrate_imu_camera --target april_6x6.yaml --cam camchain.yaml --imu imu_adis16448.yaml --bag dynamic.bag --bag-from-to 5 45

but there is an error:
`RuntimeError: [ImuConfig Reader]: Could not read configuration from imu_adis16448.yaml`

this is the log:
```
 Initializing IMUs:
 Traceback_ (most recent call last):
  File "/home/yanyan/kalibr_workspace/devel/bin/kalibr_calibrate_imu_camera", line 6, in <module>
    exec(fh.read())
  File "<string>", line 236, in <module>
  File "<string>", line 147, in main
  File "/home/yanyan/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 320, in __init__
    ParametersBase.__init__(self, yamlFile, "ImuConfig", createYaml)
  File "/home/yanyan/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 149, in __init__
    self.data = self.readYaml()
  File "/home/yanyan/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 160, in readYaml
    self.raiseError( "Could not read configuration from {0}".format(self.yamlFile) )
  File "/home/yanyan/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 183, in raiseError
    raise RuntimeError( "{0}{1}".format(header, message) )\
```

Can anyone tell me why it happened and how to solve it.

Thanks!

rehderj, 2016-10-30:
Hi,
thanks for using our toolbox! There are two possible reasons: Either the file cannot be found or it is malformed. If you followed the instructions in the wiki and downloaded the yaml there, it is likely the first. So please make sure that all files exist and all paths are correct.
Also, please close this issue if your problem could be solved.
Cheers!

closed 2016-10-31 by wyycv

SubMishMar, 2018-04-29:
I am having the same issue.. where should I keep the downloaded yaml files so that the program is able to find them?

SubMishMar, 2018-04-29:
I figured it out had to give the full path to the file..  eg:

kalibr_calibrate_imu_camera --target /home/subodh/Downloads/dynamic/april_6x6.yaml --cam /home/subodh/Downloads/dynamic/camchain.yaml --imu /home/subodh/Downloads/dynamic/imu_adis16448.yaml --bag /home/subodh/Downloads/dynamic/dynamic.bag --bag-from-to 5 45

schneith, 2018-05-02:
If you are using the CDE package that might be caused but the translation of paths; which needs a full path. The compiled version should work with relative paths.

SubMishMar, 2018-05-02:
Yeah I made it work by providing the full path.. thanks anyway :)

wojiaozhupengfei, 2019-01-09:
This problem is the same as yours. The reason is that the dynamic folder is placed directly in the workspace. You should extract the yaml file in the workspace, or dynamic should be added to the path,eg:
kalibr_calibrate_imu_camera --target dynamic/april_6x6.yaml --cam dynamic/camchain.yaml --imu dynamic/imu_adis16448.yaml --bag dynamic/dynamic.bag --bag-from-to 5 45

suho0515, 2020-04-23:
> I figured it out had to give the full path to the file.. eg:
>
> kalibr_calibrate_imu_camera --target /home/subodh/Downloads/dynamic/april_6x6.yaml --cam /home/subodh/Downloads/dynamic/camchain.yaml [...]

thanks! it worked perfectly!