ethz-asl/kalibr #82: Ubuntu 16.04 / ROS kinetic issues

ojura, 2016-11-15:
Hi, I tried to run Kalibr on Ubuntu 16.04 and ROS Kinetic and ran into the following issues:

- `Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ImageDatasetReader.py` imports cv, which does not exist anymore in OpenCV3 (it used to be imported as cv2.cv in OpenCV2), so this fails like this: 
` File "...src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ImageDatasetReader.py", line 1, in <module>`
 `   import cv`
`  File "/usr/lib/python2.7/dist-packages/cv.py", line 1, in <module>`
`    from cv2.cv import *`
`ImportError: No module named cv`

  However, OpenCV2 is still available in the Ubuntu repository as a Python module (package `python-opencv`). Python can be forced to use this version of the `cv2` module, which includes `cv2.cv`, instead of the ROS OpenCV3 one by prefixing the system module store in front of the ROS one in PYTHONPATH:
` export PYTHONPATH=/usr/lib/python2.7/dist-packages:$PYTHONPATH`
  This allows Kalibr to be executed.

- `--show-extraction` spawns windows which are for some reason not redrawn, so the desktop manager starts showing them as grayscale

- when running Kalibr, there is a lot of seemingly harmless warnings like the following
<details>
  <summary>Click to expand</summary>
importing libraries
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::cameras::NoDistortion> already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::cameras::EquidistantDistortion> already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::cameras::RadialTangentialDistortion> already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::cameras::FovDistortion> already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::cameras::PinholeProjection<aslam::cameras::NoDistortion> > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::cameras::PinholeProjection<aslam::cameras::RadialTangentialDistortion> > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::cameras::PinholeProjection<aslam::cameras::EquidistantDistortion> > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::cameras::PinholeProjection<aslam::cameras::FovDistortion> > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::cameras::OmniProjection<aslam::cameras::NoDistortion> > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::cameras::OmniProjection<aslam::cameras::RadialTangentialDistortion> > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::cameras::OmniProjection<aslam::cameras::FovDistortion> > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::cameras::GlobalShutter> already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::cameras::RollingShutter> already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::backend::DesignVariableAdapter<aslam::cameras::NoDistortion> > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::backend::DesignVariableAdapter<aslam::cameras::RadialTangentialDistortion> > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::backend::DesignVariableAdapter<aslam::cameras::EquidistantDistortion> > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::backend::DesignVariableAdapter<aslam::cameras::FovDistortion> > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::backend::DesignVariableAdapter<aslam::cameras::PinholeProjection<aslam::cameras::NoDistortion> > > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::backend::DesignVariableAdapter<aslam::cameras::PinholeProjection<aslam::cameras::RadialTangentialDistortion> > > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::backend::DesignVariableAdapter<aslam::cameras::PinholeProjection<aslam::cameras::EquidistantDistortion> > > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::backend::DesignVariableAdapter<aslam::cameras::PinholeProjection<aslam::cameras::FovDistortion> > > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::backend::DesignVariableAdapter<aslam::cameras::OmniProjection<aslam::cameras::NoDistortion> > > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::backend::DesignVariableAdapter<aslam::cameras::OmniProjection<aslam::cameras::RadialTangentialDistortion> > > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::backend::DesignVariableAdapter<aslam::cameras::OmniProjection<aslam::cameras::EquidistantDistortion> > > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::backend::DesignVariableAdapter<aslam::cameras::OmniProjection<aslam::cameras::FovDistortion> > > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::backend::DesignVariableAdapter<aslam::cameras::GlobalShutter> > already registered; second conversion method ignored.
  
/home/juraj/catkin_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py:9: RuntimeWarning: to-Python converter for boost::shared_ptr<aslam::backend::DesignVariableAdapter<aslam::cameras::RollingShutter> > already registered; second conversion method ignored.
  
Initializing IMUs:
...
</details>

ojura, 2017-01-26:
@mhkabir Unfortunately, no. I reboot into Ubuntu 14.04/Indigo when I have to use Kalibr.

BTW, #75 fixed bullet 1 in this issue.

russellaabuchanan, 2017-03-28:
I'm having a different problem on Ubuntu 16.04/Kinetic (not sure if I should create another issue). I can get Kalibr to build but when I run kalibr_calibrate_cameras I get the following error:

importing libraries
Traceback (most recent call last):
  File "/home/raab/kalibr_workspace/devel/bin/kalibr_calibrate_cameras", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/raab/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 6, in <module>
    import sm
  File "/home/raab/kalibr_workspace/devel/lib/python2.7/dist-packages/sm/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 4, in <module>
ImportError: /home/raab/kalibr_workspace/devel/.private/sm_logging/lib/libsm_logging.so: undefined symbol: _ZN5boost13match_resultsIN9__gnu_cxx17__normal_iteratorIPKcSsEESaINS_9sub_matchIS5_EEEE12maybe_assignERKS9_

raabuchanan, 2018-01-22:
@DennisMelamed Unfortunately no, I could only get it running on Ubuntu 14.

DennisMelamed, 2018-01-24:
@raabuchanan I think I got it figured out, ensure that the version of boost you are using is compiled with the same gcc/g++ version as the one you are using to build Kalibr. I was using gcc4.8 to build Kalibr but boost was built with the default gcc5.

DodsGTH94, 2018-10-18:
Ok I solved the problem.
It was just a matter of installing the latest library. So first i did `sudo apt-get install python-igraph` but it didn´t work.
Then here there are the instructions I executed:
`sudo add-apt-repository ppa:igraph/ppa
sudo apt-get update
sudo apt-get install python-igraph`
Hope it will be useful.

ytzhl, 2020-04-05:
exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/z/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_create_target_pdf", line 5, in <module>
    from pyx import *
ImportError: No module named pyx

xiaobrnbrn, 2020-08-26:
> I'm having a different problem on Ubuntu 16.04/Kinetic (not sure if I should create another issue). I can get Kalibr to build but when I run kalibr_calibrate_cameras I get the following error:
> 
> importing libraries
> Traceback (most recent call last):
> File "/home/raab/kalibr_workspace/devel/bin/kalibr_calibrate_cameras", line 15, in
> exec(compile(fh.read(), python_script, 'exec'), context)
> File "/home/raab/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 6, in
> import sm
> File "/home/raab/kalibr_workspace/devel/lib/python2.7/dist-packages/sm/**init**.py", line 35, in
> exec(__fh.read())
> File "", line 4, in
> ImportError: /home/raab/kalibr_workspace/devel/.private/sm_logging/lib/libsm_logging.so: undefined symbol: _ZN5boost13match_resultsIN9__gnu_cxx17__normal_iteratorIPKcSsEESaINS_9sub_matchIS5_EEEE12maybe_assignERKS9_

Hello,I met the same problem, did you solve this problem?

closed 2022-05-03 by goldbattle