ethz-asl/kalibr #84: Problem when installed

sunbinbin1991, 2016-12-12:
Hi, I have a problem when I use this tool to calibrate,here is error:

kalibr-cde ➤ ./kalibr_bagcreater --folder ../binbin/calibrate_data_7/output --output-bag bin.bag            
importing libraries
ImportError: numpy.core.multiarray failed to import
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils

Original exception was:
ImportError: numpy.core.multiarray failed to import
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_bagcreater", line 5, in <module>
    exec(fh.read())
  File "<string>", line 11, in <module>
ImportError: numpy.core.multiarray failed to import
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils

Original exception was:
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_bagcreater", line 5, in <module>
    exec(fh.read())
  File "<string>", line 11, in <module>
ImportError: numpy.core.multiarray failed to import

SO,what could be reason for this ? Actually, several days ago, it's ok, when I install others soft, then I got this problem, my system is ubuntu 15.04

closed 2017-07-20 by sunbinbin1991

halajun, 2017-08-06:
@sunbinbin1991 how did you solve this problem?

sunbinbin1991, 2017-08-07:
I have solved this problem, but  i forget it in details

inecho, 2018-07-25:
i have the same problem. can anyone help me out?

alreadyyang, 2019-11-22:
I have the same problem,who can help me ?
Thx

sunbinbin1991, 2019-11-22:
@alreadyyang   maybe your numpy version is not right, you can try the solution behind:
`pip install -U numpy`
https://github.com/tensorflow/tensorflow/issues/559