ethz-asl/kalibr #85: Build problem with Ubuntu 15.10 and boost 1.58

rikpires, 2016-12-15:
Hi,

I have a problem building the project. The error finally locates at "error: invalid conversion from ‘long unsigned int’ to ‘void*’ " in boost/archive/detail/iserializer.hpp. I have searched but no solution found yet. Does anyone meet this error before? Thanks!

- Ubuntu version: 15.10
- boost version: 1.58.0

> In file included from /usr/include/boost/archive/detail/interface_iarchive.hpp:22:0,
                 from /usr/include/boost/archive/detail/common_iarchive.hpp:23,
                 from /usr/include/boost/archive/basic_text_iarchive.hpp:31,
                 from /usr/include/boost/archive/text_iarchive.hpp:24,
                 from /home/yanbin/code/calibration/kalibr_workspace/src/kalibr/Schweizer-Messer/sm_boost/include/sm/boost/serialization.hpp:6,
                 from /home/yanbin/code/calibration/kalibr_workspace/src/kalibr/aslam_cv/aslam_cameras/include/aslam/cameras/CameraGeometryBase.hpp:8,
                 from /home/yanbin/code/calibration/kalibr_workspace/src/kalibr/aslam_cv/aslam_cameras/include/aslam/cameras/CameraGeometry.hpp:4,
                 from /home/yanbin/code/calibration/kalibr_workspace/src/kalibr/aslam_cv/aslam_cameras/include/aslam/cameras.hpp:5,
                 from /home/yanbin/code/calibration/kalibr_workspace/src/kalibr/aslam_cv/aslam_cv_serialization/include/aslam/cameras/CameraBaseSerialization.hpp:15,
                 from /home/yanbin/code/calibration/kalibr_workspace/src/kalibr/aslam_cv/aslam_cv_serialization/src/autogen/Camera-PinholeCameraGeometry.cpp:1:
/usr/include/boost/archive/detail/iserializer.hpp: In instantiation of ‘static void boost::archive::detail::heap_allocation<T>::has_new_operator::invoke_delete(T*) [with T = aslam::cameras::CameraGeometry<aslam::cameras::PinholeProjection<aslam::cameras::NoDistortion>, aslam::cameras::GlobalShutter, aslam::cameras::NoMask>:
/usr/include/boost/archive/detail/iserializer.hpp:265:33:   required from ‘static void boost::archive::detail::heap_allocation<T>::invoke_delete(T*) [with T = aslam::cameras::CameraGeometry<aslam::cameras::PinholeProjection<aslam::cameras::NoDistortion>, aslam::cameras::GlobalShutter, aslam::cameras::NoMask>
/usr/include/boost/archive/detail/iserializer.hpp:273:26:   required from ‘boost::archive::detail::heap_allocation<T>::~heap_allocation() [with T = aslam::cameras::CameraGeometry<aslam::cameras::PinholeProjection<aslam::cameras::NoDistortion>, aslam::cameras::GlobalShutter, aslam::cameras::NoMask>
/usr/include/boost/archive/detail/iserializer.hpp:294:36:   required from ‘void* boost::archive::detail::pointer_iserializer<Archive, T>::heap_allocation() const [with Archive = boost::archive::portable_binary_iarchive; T = aslam::cameras::CameraGeometry<aslam::cameras::PinholeProjection<aslam::cameras::NoDistortion>, aslam::cameras::GlobalShutter, aslam::cameras::NoMask>
/home/yanbin/code/calibration/kalibr_workspace/src/kalibr/aslam_cv/aslam_cv_serialization/src/autogen/Camera-PinholeCameraGeometry.cpp:14:68:   required from here
/usr/include/boost/archive/detail/iserializer.hpp:237:47: error: invalid conversion from ‘long unsigned int’ to ‘void*’ [-fpermissive]
                 (T::operator delete)(t, sizeof(T));
                                               ^

rpng-guest, 2018-07-03:
Hi, did you solve this problem? I met this too.

drscopus, 2018-09-18:
I met this problem too.