ethz-asl/kalibr #86: a little confused about the gravity vector

maoyuyuyuyuyu, 2017-01-13:
hi
I just tried the example dataset to run the Camera IMU calibration。
In  the dataset ，the datas showed at the beginning：
timestamp,    omega_x,     omega_y,   omega_z,   alpha_x,  alpha_y,   alpha_z
1402065930415759104,-0.0055850536063818549,-0.0027925268031909274,0.017453292519943295,9.733100125,-0.62108783333333339,-0.46581587499999999
1402065930420759040,-0.0013962634015954637,-0.0020943951023931952,0.024434609527920613,9.7249279166666671,-0.67829329166666663,-0.4494714583333333
1402065930425758976,-0.0048869219055841231,-0.00069813170079773186,0.02652900463031381,9.7984777916666665,-0.83356525000000004,-0.44129924999999998
the value of alpha_x is very closed to the value of gravity acceleration. According to this, I first assume that the X-axis represents the direction of the gravity in the coordinate of Imu.
The first Question is , for this Kalibr , is that important to fix the X-axis  to the direction of Gravity? 

And the results report show that:
Gravity vector in target coords: [m/s^2]
[-0.06523734 8.78297289 -4.36159995]
Is the target coordinate represents the world coordinate? And is that means that Y-axis represents the direction of the gravity in the world coordinate of Kalibr?

Thanks !!

hhmaizi, 2017-08-30:
I guess it means your chessboard's Y direction is not aligned with gravity. this vector is your chessboard's direction.

chunibyo-wly, 2021-01-24:
Hello, do you fix this yet?
What should I suppose to get (`gravity vector in target coords`) when I put an april tag in this orientation
[image]

`axis Y` is up
`axis X` is right
`axis Z` is to person

I think the `gravity vector.y` is supposed to be around `-9.8`, am I right?

thanks for your reply.