ethz-asl/kalibr #87: Is constant time interval and perfect alignment necessary for IMU and Camera input?

liyinnb, 2017-01-19:
I noticed that the timestamps of IMU readings and camera frames in the dynamic sample are perfectly evenly distributed. This means that the camera frame arrives exactly at 20Hz and IMU reading arrives exactly at 200Hz. May I know is this a necessary requirement for good calibration result? Also, I noticed that camera frames arrive exactly at the time when an IMU reading is available. This happens when the camera frame interval is divisible by the IMU reading interval. However, what if my camera frame rate is 30Hz and IMU reading is 100Hz?

ZacharyTaylor, 2017-01-19:
Kalibr uses a continuous time model for the motion of the sensors. This means that the times at which a sensor obtains readings does not need to be equally spaced. It also means the readings of multiple sensors do not need to coincide.

All that is needed timing wise is that the timestamp information for the sensors is accurate. Kalibr can also handle a consistent offset in timing between different sensors by using the "--time-calibration" flag.

liyinnb, 2017-01-20:
Great! @ZacharyTaylor thanks for the clarification :)

closed 2017-01-20 by liyinnb