ethz-asl/kalibr #88: kalibr_calibrate_cameras gets stuck

hemanthkorrapati, 2017-01-26:
I am trying to calibrate multiple cameras together. No matter what parameters and how ever many images i use, it always stops after processing 7 images and just waits there indefinitely. 
This is the output I got:

`Initializing cam0:
	Camera model:	  pinhole-fov
	Dataset:          calib_images.bag
	Topic:            s1_left
	Number of images: 67
Extracting calibration target corners
  Progress 7 / 67 	 Time remaining: 6s `

After this point, it does not progress for ever. 

Did anyone face the same issue ?  (I am using ros-kinetic on ubuntu-16.04.)

MichaelGrupp, 2017-05-10:
Maybe this helps: check if there is a blocking window with the title "Wild April-Tag detected! ..." and (force) close it.

melodybinbin, 2019-03-15:
@hemanthkorrapati Hi, do you solve your problem? I come with the same problem? I will appreciate for your help. Thanks in advance.

MohrMa2, 2019-09-13:
No I was wrong. It wasn't a problem with size, but with wild apriltags. The calibration pattern was mirrored in a glassy surface. When apriltags are detected, the cc code quits, but in python it looks like it hangs. (Reducing the image size accidently altered the  wild aprilgrids, so they were not detected anymore).

mintar, 2019-11-05:
I ran into the same problem and can confirm it's caused by a "bad" image where the calibration pattern was incorrectly detected. The solution was to use `kalibr_bagextractor` to extract the bagfile to image folders, then delete the image where it hangs, and then use `kalibr_bagcreater` to convert the images back into a bag file.

closed 2022-04-22 by goldbattle