ethz-asl/kalibr #89: bagcreater doesn't write camera images in the CDE download

mklingen, 2017-01-26:
Since I cannot install ROS on my system I decided to try to run your framework using the CDE package in the Downloads section. I am running Ubuntu 14.04. I created a dataset in the format specified in the wiki for "bagcreater" (btw it should be spelled "bagcreator"), and ran the command:

./kalibr_bagcreater --folder ~/my_dataset_dir/ --output-bag ~/my_dataset_dir/my_dataset.bag

The resulting bag file is only a few megabytes and doesn't seem to contain any camera data. What could be going on?

EDIT: turns out the path to the folder can't contain a slash at the end!

rehearsal-, 2017-02-10:
@mklingen hi, I also met this problem, and after I changed to an absolute path, as you said, it still creates a few negabytes bag file. BTW, I arrange all needed files follow the instruction [here](https://github.com/ethz-asl/kalibr/wiki/bag-format)

 Is there anything I missed? Thanks in advance.

liming0791, 2017-03-17:
I have the same problem, and an absolute path to the dataset has no effect on the result. I have no idea what is wrong.

mklingen, 2017-03-17:
you need a dot "." at the end of the absolute path.

liming0791, 2017-03-20:
@mklingen It works. Thanks a lot !

sqn175, 2017-06-06:
@mklingen Hi, I add a dot "." at the end of my absolute path, but the rosbag contains no image topics.
the files is as follows,
datasetdir
├── cam0
│   ├── 1234567890123456789.png
│   └── 1234567890133456789.png
└── imu0.csv
and I ran the command:
./kalibr_bagcreater --folder /home/sqn/Documents/dataserdir/. --output-bag mybag.bag
The result bag contains the correct IMU topic, but no images.  
Is there anything wrong?

HTLife, 2018-02-19:
@mklingen What do you mean by 
>you need a dot "." at the end of the absolute path.
Could you provide some example?

HTLife, 2018-02-19:
The example is provided below:
```bash
kalibr_bagcreater --folder /home/user/code/v2_01/. --output-bag awsome.bag
```
Should this be added to wiki for clarification?

https://github.com/ethz-asl/kalibr/wiki/bag-format

goldbattle, 2022-04-22:
Wiki has been edited, will track an actual fix via issue https://github.com/ethz-asl/kalibr/issues/146

closed 2022-04-22 by goldbattle