ethz-asl/kalibr #92: Cannot generate calibration target with master source

nhanph, 2017-02-22:
Hi,

I try to use the script to generate the calibration target for camera calibration and I get this error:

```
Traceback (most recent call last):
  File "/home/nhanph/kalibr_workspace/devel/bin/kalibr_create_target_pdf", line 6, in <module>
    exec(fh.read())
  File "<string>", line 5, in <module>
ImportError: No module named pyx
```

Could you help me with this?

Thanks in advance.

schneith, 2017-02-22:
You are missing the pyx dependency. On Ubuntu, you can install it with:
```
sudo apt-get install python-pyx
```

nhanph, 2017-02-23:
It works fine now. Thank you very much.

closed 2017-02-23 by nhanph

gaozhiye, 2019-12-07:
> You are missing the pyx dependency. On Ubuntu, you can install it with:
> 
> ```
> sudo apt-get install python-pyx
> ```

Hi!
I try to generate the calibration board .When I run the commond line ,
kalibr_create_target_pdf --type apriltag --nx 6 --ny 6 --tsize 0.088 --tspace 0.3
the reply is , commond not found .
Can you give me a calibration file in PDF format ?
Thank you very much !

Pratiquea, 2021-10-04:
> You are missing the pyx dependency. On Ubuntu, you can install it with:
> 
> ```
> sudo apt-get install python-pyx
> ```

I had issues installing pyx both via apt (Error: _Unable to locate package python-pyx_) and pip (Error: _ERROR: No .egg-info directory found in /tmp/pip-pip-egg-info-uO0AEb_)
Resolved the issue by installing a specific version via pip
```
sudo -H pip install PyX==0.12
```