ethz-asl/kalibr #97: The acc errors appear to be all zero.

Userskii, 2017-04-26:
I did a combined VI sensor calibration using the Camera-IMU calibration tool. However the gyro and acc errors are zero somehow. This is shown like 

`Calibration results
===================
Normalized Residuals
----------------------------
Reprojection error (cam0):     mean 0.161373177542, median 0.143201264718, std: 0.100674912244
Reprojection error (cam1):     mean 0.16662511419, median 0.146514104122, std: 0.104168779993
Gyroscope error (imu0):        mean 3.72979172913e-10, median 2.24605333102e-10, std: 1.22841183962e-09
Accelerometer error (imu0):    mean 8.98662346209e-10, median 4.43878332656e-11, std: 1.53928329578e-08`

What could be the possible cause for this?

[imu0.txt](https://github.com/ethz-asl/kalibr/files/957551/imu0.txt)

Userskii, 2017-05-03:
The imu data is not collected correctly.

closed 2017-05-03 by Userskii

bdjohnson529, 2018-01-05:
I am having the same issue. How did you correct your setup to collect IMU data properly?

MerlotQ, 2018-01-05:
@bdjohnson529 make sure that the sampling rate is correct