ethz-asl/kalibr #99: ERROR:  initialization of focal length

einspyon, 2017-05-04:
Extracted corners for 388 images (of 519 images)                              
[ERROR] [1493897823.130380]: initialization of focal length for cam with topic /cam0/image_raw failed  
	Projection initialized to: [ nan  nan  nan  nan]
	Distortion initialized to: [ nan  nan  nan  nan]

why? beg for help

skohlbr, 2017-05-06:
I ran into the same problem and implemented a hack that just generates a guess if kalibr wasn't able to generate an inital guess, see here: https://github.com/tu-darmstadt-ros-pkg/kalibr/commit/2822f6a6f47b92b600de42fd1e82915f1b0b2fd7

This approach worked fine when calibrating a multi-camera system, even though I just put in 400 as a magic number for the focal length initial estimate.

Doodleshr, 2017-06-13:
same error here, does anyone know how to solve it under CDE?

utkarsh447, 2019-02-08:
> same error here, does anyone know how to solve it under CDE?

Hi, Did anyone fine solution for CDE? As the file mentioned in this commit [tu-darmstadt-ros-pkg@2822f6a](https://github.com/tu-darmstadt-ros-pkg/kalibr/commit/2822f6a6f47b92b600de42fd1e82915f1b0b2fd7)  is not present in kalibr-cde.

LarryDong, 2021-08-08:
I tried to solve this problem for CDE version by manually give an initial guess. Please check [issue-367](https://github.com/ethz-asl/kalibr/issues/367#issuecomment-846945763)

I set cam0's distortion to all zeros and the projection estimated by its focal length & resolution, like this:
```bash
cameraList[0].geometry.projection().setParameters(np.array(1600,1706,640,512]))
cameraList[0].geometry.projection().distortion().setParameters(np.array([0,0,0,0]))
```
Since my sensor is 1/3'' inch, with a resolution of 1280x1024, and focus length should be 6mm. So 1600 is calculated by: 6mm/4.8mmx1280, 1706=6/3.6x1024, and 640&512 is half the resolution.

The final result from kalibr is: [2094.6, 2114.3, 497.3, 464.8], which seems reasonable.

Hope it helps.

Horizon-42, 2021-09-24:
> > > I tried to solve this problem for CDE version by manually give an initial guess. Please check [issue-367](https://github.com/ethz-asl/kalibr/issues/367#issuecomment-846945763)
> > > I set cam0's distortion to all zeros and the projection estimated by its focal length & resolution, like this:
> > > ```shell
> > > cameraList[0].geometry.projection().setParameters(np.array(1600,1706,640,512]))
> > > cameraList[0].geometry.projection().distortion().setParameters(np.array([0,0,0,0]))
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > Since my sensor is 1/3'' inch, with a resolution of 1280x1024, and focus length should be 6mm. So 1600 is calculated by: 6mm/4.8mmx1280, 1706=6/3.6x1024, and 640&512 is half the resolution.
> > > The final result from kalibr is: [2094.6, 2114.3, 497.3, 464.8], which seems reasonable.
> > > Hope it helps.
> > 
> > 
> > can you tell me which line you changed in the cde file? I insert this two line in the main while loop, but it did not work.
> > thank you!!!
> 
> Yes I put those line just in the while true loop.
> `kalibr_calibrate_cameras` file, `main()` function, `while true:` loop. I guess the line number is 208 here.

thanks!

goldbattle, 2022-04-22:
Should be fixed via manual specification: https://github.com/ethz-asl/kalibr/pull/346

closed 2022-04-22 by goldbattle