ethz-asl/kalibr #103: Camera calibration reprojection error

acajic, 2017-05-21:
In some places in the wiki pages it is written that a good camera calibration is the one having the reprojection error between 0.1 - 0.2 px. 

For what resolution? It seems natural that this must matter.
I am getting 0.2 - 0.4 px for 2048x1088, how good/bad is that?

ashish-kb, 2017-05-26:
I have this doubt too, could someone please clarify. I am confused on what do these reprojection errors signify? Are they considering only intrinsic or do they include the imu to camera to tag reprojection?

hitimo, 2017-05-26:
I think the reproj. error of 0.1-0.2 px referred to low-resolution (such as the 0.3 MP Aptina MT9v034 of the VI-Sensor) cameras. The reprojection error returned by Kalibr does not take the camera intrinsics into account:
https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/CameraCalibrator.py#L393-L395

caselitz, 2017-05-29:
Shouldn't this somehow normalize out (meaning it is resolution independent)?

Not sure if that makes sense but the idea is the following:

For lower resolutions the calibration is less accurate since checkerboard corners are less precise due to pixel discretization. The worse calibration causes a higher reprojection error in meters, but this is counter-weight by the fact that 1 pixel also corresponds to a larger distance in meters for low resolutions.

closed 2022-04-22 by goldbattle