ethz-asl/kalibr #106: 【Build error】28 packages were abandoned

sqn175, 2017-06-15:
Hi, 
I am trying to compile Kalibr from source follow the [steps](https://github.com/ethz-asl/kalibr/wiki/installation#b-building-from-source), but I'm getting these errors:

```
Profile:                     default
Extending:        [explicit] /opt/ros/indigo
Workspace:                   /home/sqn/kalibr_workspace
--------------------------------------------------------------------------------
Source Space:       [exists] /home/sqn/kalibr_workspace/src
Log Space:          [exists] /home/sqn/kalibr_workspace/logs
Build Space:        [exists] /home/sqn/kalibr_workspace/build
Devel Space:        [exists] /home/sqn/kalibr_workspace/devel
Install Space:      [unused] /home/sqn/kalibr_workspace/install
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        -j4
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------------
[build] Found '37' packages in 0.0 seconds.                                    
[build] Package table is up to date.                                           
Starting  >>> catkin_simple                                                    
Starting  >>> ethz_apriltag2                                                   
Starting  >>> python_module                                                    
Starting  >>> sm_common                                                        
Finished  <<< sm_common                                     [ 0.1 seconds ]    
Starting  >>> sm_logging                                                       
Finished  <<< catkin_simple                                 [ 0.1 seconds ]    
Starting  >>> sm_random                                                        
Finished  <<< python_module                                 [ 0.1 seconds ]    
Starting  >>> suitesparse                                                      
Finished  <<< sm_logging                                    [ 0.1 seconds ]    
Starting  >>> sm_boost                                                         
Finished  <<< sm_random                                     [ 0.1 seconds ]    
Starting  >>> sm_matrix_archive                                                
_______________________________________________________________________________
Errors     << ethz_apriltag2:make /home/sqn/kalibr_workspace/logs/ethz_apriltag2/build.make.001.log
CMakeFiles/apriltags_demo.dir/src/example/apriltags_demo.cpp.o: In function `Demo::loop()':
apriltags_demo.cpp:(.text._ZN4Demo4loopEv[_ZN4Demo4loopEv]+0x179): undefined reference to `cv::_OutputArray::_OutputArray(cv::Mat&)'
apriltags_demo.cpp:(.text._ZN4Demo4loopEv[_ZN4Demo4loopEv]+0x188): undefined reference to `cv::_InputArray::_InputArray(cv::Mat const&)'
apriltags_demo.cpp:(.text._ZN4Demo4loopEv[_ZN4Demo4loopEv]+0x2fe): undefined reference to `cv::_InputArray::_InputArray(cv::Mat const&)'
apriltags_demo.cpp:(.text._ZN4Demo4loopEv[_ZN4Demo4loopEv]+0x324): undefined reference to `cv::imshow(std::string const&, cv::_InputArray const&)'
CMakeFiles/apriltags_demo.dir/src/example/apriltags_demo.cpp.o: In function `Demo::setup()':
apriltags_demo.cpp:(.text._ZN4Demo5setupEv[_ZN4Demo5setupEv]+0x321): undefined reference to `cv::VideoCapture::get(int)'
apriltags_demo.cpp:(.text._ZN4Demo5setupEv[_ZN4Demo5setupEv]+0x333): undefined reference to `cv::VideoCapture::get(int)'
apriltags_demo.cpp:(.text._ZN4Demo5setupEv[_ZN4Demo5setupEv]+0x585): undefined reference to `cv::namedWindow(std::string const&, int)'
apriltags_demo.cpp:(.text._ZN4Demo5setupEv[_ZN4Demo5setupEv]+0x615): undefined reference to `cv::Ptr<CvCapture>::delete_obj()'
/home/sqn/kalibr_workspace/devel/.private/ethz_apriltag2/lib/libethz_apriltag2.so: undefined reference to `cv::circle(cv::Mat&, cv::Point_<int>, int, cv::Scalar_<double> const&, int, int, int)'
/home/sqn/kalibr_workspace/devel/.private/ethz_apriltag2/lib/libethz_apriltag2.so: undefined reference to `cv::line(cv::Mat&, cv::Point_<int>, cv::Point_<int>, cv::Scalar_<double> const&, int, int, int)'
/home/sqn/kalibr_workspace/devel/.private/ethz_apriltag2/lib/libethz_apriltag2.so: undefined reference to `cv::putText(cv::Mat&, std::string const&, cv::Point_<int>, int, double, cv::Scalar_<double>, int, int, bool)'
/home/sqn/kalibr_workspace/devel/.private/ethz_apriltag2/lib/libethz_apriltag2.so: undefined reference to `vtable for cv::_OutputArray'
/home/sqn/kalibr_workspace/devel/.private/ethz_apriltag2/lib/libethz_apriltag2.so: undefined reference to `cv::findHomography(cv::_InputArray const&, cv::_InputArray const&, int, double, cv::_OutputArray const&)'
/home/sqn/kalibr_workspace/devel/.private/ethz_apriltag2/lib/libethz_apriltag2.so: undefined reference to `vtable for cv::_InputArray'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/sqn/kalibr_workspace/devel/.private/ethz_apriltag2/lib/ethz_apriltag2/apriltags_demo] Error 1
make[1]: *** [CMakeFiles/apriltags_demo.dir/all] Error 2
make: *** [all] Error 2
cd /home/sqn/kalibr_workspace/build/ethz_apriltag2; catkin build --get-env ethz_apriltag2 | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << ethz_apriltag2:make                           [ Exited with code 2 ]
[... 24 lines cut ...]
Abandoned <<< bsplines_python                               [ Unrelated job failed ]
Abandoned <<< aslam_splines_python                          [ Unrelated job failed ]
Abandoned <<< incremental_calibration                       [ Unrelated job failed ]
Abandoned <<< incremental_calibration_python                [ Unrelated job failed ]
Abandoned <<< kalibr                                        [ Unrelated job failed ]
Finished  <<< suitesparse                                   [ 0.1 seconds ]    
Finished  <<< sm_matrix_archive                             [ 3.5 seconds ]     
Finished  <<< sm_boost                                      [ 3.9 seconds ]    
[build] Summary: 8 of 37 packages succeeded.                                   
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 28 packages were abandoned.                               
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 4.5 seconds total.                                            
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

I'm working on Ubuntu 14.04 with ROS indigo. 
And I have installed two OpenCV version, one by using the command:
` sudo apt-get install libopencv-dev`
one by installing from source.

How can I get rid of these errors?

Thanks in advance

fmina, 2017-06-15:
You have to change your devel space layout from `linked` to `merged`. Check catkin tools documentation.

sqn175, 2017-06-15:
Hi, @fmina 

Thanks for your reply. 

I changed the devel space using command:
`catkin config --merge-devel`

and got:
```
----------------------------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/indigo
Workspace:                   /home/sqn/kalibr_workspace
----------------------------------------------------------------------------------
Source Space:       [exists] /home/sqn/kalibr_workspace/src
Log Space:         [missing] /home/sqn/kalibr_workspace/logs
Build Space:        [exists] /home/sqn/kalibr_workspace/build
Devel Space:        [exists] /home/sqn/kalibr_workspace/devel
Install Space:      [unused] /home/sqn/kalibr_workspace/install
DESTDIR:            [unused] None
----------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
----------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        -j4
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
----------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
```
And I start a fresh build, but the same error occurs and 30 packages were abandoned. 

Is there something wrong?

fmina, 2017-06-15:
you have to clean your workspace with `catkin clean --all` then rebuild.

sqn175, 2017-06-15:
Yes, I do clean the workspace using the command, and it does not work.

I finally build it successfully by uninstalling OpenCV 3.0.0 libraries.

Anyway, Thanks!

closed 2017-06-15 by sqn175