ethz-asl/kalibr #107: bagcreater import error

MF1523017, 2017-06-18:
> --output-bag test.bag
importing libraries
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_bagcreater", line 5, in <module>
    exec(fh.read())
  File "<string>", line 121, in <module>
  File "<string>", line 87, in createImuMessge
ValueError: invalid literal for int() with base 10: ''
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils
Original exception was:
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_bagcreater", line 5, in <module>
    exec(fh.read())
  File "<string>", line 121, in <module>
  File "<string>", line 87, in createImuMessge
ValueError: invalid literal for int() with base 10:

does someone have this problem?

closed 2017-06-19 by MF1523017

phyorch, 2018-04-24:
Hei, How about your solution at last?

mengdong824, 2019-05-14:
I met the same error, do you solve this? Thx

ruikunl, 2019-05-28:
This is the problem with your file name for the images. 

In the kalibr_bagcreator.py file, it gets the timestep by:

`    timestamp_nsecs = os.path.splitext(os.path.basename(filename))[0]
    timestamp = rospy.Time( secs=int(timestamp_nsecs[0:-9]), nsecs=int(timestamp_nsecs[-9:]) )`

which means that if your file name is too short, the nsecs will try to convert an empty string to int. 

Also check this [https://github.com/ethz-asl/kalibr/issues/190](url), which says your file name should be the timestep.