ethz-asl/kalibr #110: Initialization of focal length fails when calibrating with pinhole-radtan model

Nurgak, 2017-06-27:
I'm trying to use kalibr to calibrate a multi-camera system, but I keep getting an error right at the beginning:

`kalibr_calibrate_cameras --bag bb2_bb3_pancam.bag --topics /cam5/image_raw /cam6/image_raw --models pinhole-radtan pinhole-radtan --target aprilgrid_target_6x6.yaml`

This results in the following error:

```
Initializing cam0:
	Camera model:	  pinhole-radtan
	Dataset:          bb2_bb3_pancam.bag
	Topic:            /cam5/image_raw
	Number of images: 181
Extracting calibration target corners
  Extracted corners for 115 images (of 181 images)                              
[ERROR] [1498566878.788810]: initialization of focal length for cam with topic /cam5/image_raw failed  
Traceback (most recent call last):
  File "/home/marta/catkin_ws/devel/bin/kalibr_calibrate_cameras", line 6, in <module>
    exec(fh.read())
  File "<string>", line 444, in <module>
  File "<string>", line 182, in main
  File "/home/marta/catkin_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/CameraCalibrator.py", line 68, in initGeometryFromObservations
    success = kcc.calibrateIntrinsics(self, observations)
  File "/home/marta/catkin_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/CameraIntializers.py", line 221, in calibrateIntrinsics
    target_pose_dv = addPoseDesignVariable(problem, T_t_c)
  File "/home/marta/catkin_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/CameraIntializers.py", line 7, in addPoseDesignVariable
    q_Dv = aopt.RotationQuaternionDv( T0.q() )
RuntimeError: [std::runtime_error] /home/marta/catkin_ws/src/kalibr/Schweizer-Messer/sm_kinematics/src/quaternion_algebra.cpp:79: quat2r() debug assert(q.norm() == 1.f) failed [-nan == 1 (nan > 0.0001)]: The quaternion must be a unit vector to represent a rotation
```

I have multiple calibration sets with the same cameras, done on different days under different lighting conditions, however one particular set keeps giving these errors which is quite unexpected and I cannot figure out why. I've tried to make sure the target is properly visible on the images, but no matter what I keep getting the quaternion issue. Has anybody encountered this?

When I use `omni-radtan` model it works well, but the intrinsics are completely off compared to when I use `pinhole-radtan` model (on the sets where the calibration works well). So I'd like it to work with the right camera model (`pinhole-radtan`).

Nurgak, 2017-06-27:
To reproduce the error use the following files, there is only one picture, but it works just as well. Call `./calibrate.sh`, this will launch the bag creation (the `.bag` file is also included) and then the kalibr calibration runs, or just copy-paste the contents of the `./calibrate.sh` into the terminal.

[kalibr_error_showcase.zip](https://github.com/ethz-asl/kalibr/files/1105536/kalibr_error_showcase.zip)

When `pinhole-radtan` is changed to `omni-radtan` it works.

veintiocho, 2017-08-12:
Hello! I met the same error as you did, have you solved it?

skohlbr, 2017-08-12:
As already mentioned in #99 , I was able to work around this by initializing the focal length to some fixed number if a guess couldn't be computed beforehand (https://github.com/tu-darmstadt-ros-pkg/kalibr/commit/2822f6a6f47b92b600de42fd1e82915f1b0b2fd7). This might not be the most elegant thing to do, but it made calibration work for me and I didn't see negative side-effects so far.

closed 2022-06-14 by goldbattle

dieumerci7, 2024-12-17:
For a record, an error with the same log might occur if you feed bad target parameters to the algorithm. In my case, I used an old .yaml file with wrong column number and tag size.