ethz-asl/kalibr #112: IndexError: index out of bounds

betterhalfwzm, 2017-07-11:
TypeError: Conversion is only valid for arrays with 1 or 2 dimensions. Argument has 3 dimensions
TypeError: Conversion is only valid for arrays with 1 or 2 dimensions. Argument has 3 dimensions
TypeError: Conversion is only valid for arrays with 1 or 2 dimensions. Argument has 3 dimensions
    success, obs = detector.findTarget(stamp, np.array(image))
    success, obs = detector.findTarget(stamp, np.array(image))
TypeError: Conversion is only valid for arrays with 1 or 2 dimensions. Argument has 3 dimensions
TypeError: Conversion is only valid for arrays with 1 or 2 dimensions. Argument has 3 dimensions
    success, obs = detector.findTarget(stamp, np.array(image))
TypeError: Conversion is only valid for arrays with 1 or 2 dimensions. Argument has 3 dimensions
    success, obs = detector.findTarget(stamp, np.array(image))
TypeError: Conversion is only valid for arrays with 1 or 2 dimensions. Argument has 3 dimensions
 
[FATAL] [1499760263.706400]: No corners could be extracted for camera /cam0/image_raw! Check the calibration target configuration and dataset.
Initializing camera rosbag dataset reader:
	Dataset:          dynamic.bag
	Topic:            /cam1/image_raw
[ WARN] [1499760266.991976]: BagImageDatasetReader: truncated 216 / 1012 images.
	Number of images: 1012
Extracting calibration target corners
Process Process-12:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/wzm/catkin_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/TargetExtractor.py", line 24, in multicoreExtractionWrapper
    success, obs = detector.findTarget(stamp, np.array(image))
TypeError: Conversion is only valid for arrays with 1 or 2 dimensions. Argument has 3 dimensions
Process Process-13:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/wzm/catkin_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/TargetExtractor.py", line 24, in multicoreExtractionWrapper
    success, obs = detector.findTarget(stamp, np.array(image))
TypeError: Conversion is only valid for arrays with 1 or 2 dimensions. Argument has 3 dimensions
Process Process-14:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/wzm/catkin_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/TargetExtractor.py", line 24, in multicoreExtractionWrapper
    success, obs = detector.findTarget(stamp, np.array(image))
TypeError: Conversion is only valid for arrays with 1 or 2 dimensions. Argument has 3 dimensions
Process Process-15:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/wzm/catkin_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/TargetExtractor.py", line 24, in multicoreExtractionWrapper
    success, obs = detector.findTarget(stamp, np.array(image))
TypeError: Conversion is only valid for arrays with 1 or 2 dimensions. Argument has 3 dimensions
Process Process-16:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/wzm/catkin_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/TargetExtractor.py", line 24, in multicoreExtractionWrapper
    success, obs = detector.findTarget(stamp, np.array(image))
TypeError: Conversion is only valid for arrays with 1 or 2 dimensions. Argument has 3 dimensions
  Progress 6 / 796 	 Time remaining: 1m 14s                 Process Process-17:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/wzm/catkin_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/TargetExtractor.py", line 24, in multicoreExtractionWrapper
    success, obs = detector.findTarget(stamp, np.array(image))
TypeError: Conversion is only valid for arrays with 1 or 2 dimensions. Argument has 3 dimensions
Process Process-18:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/wzm/catkin_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/TargetExtractor.py", line 24, in multicoreExtractionWrapper
    success, obs = detector.findTarget(stamp, np.array(image))
TypeError: Conversion is only valid for arrays with 1 or 2 dimensions. Argument has 3 dimensions
 
[FATAL] [1499760268.176857]: No corners could be extracted for camera /cam1/image_raw! Check the calibration target configuration and dataset.
Baseline between cam0 and cam1 set to:
T=  [[ 0.99999877  0.00118911 -0.00102243 -0.1101676 ]
 [-0.00118838  0.99999904  0.00071255 -0.00032166]
 [ 0.00102327 -0.00071134  0.99999922  0.00012079]
 [ 0.          0.          0.          1.        ]]
Baseline:  0.110168134052  [m]

Building the problem
	Spline order: 6
	Pose knots per second: 100
	Do pose motion regularization: False
		xddot translation variance: 1000000.000000
		xddot rotation variance: 100000.000000
	Bias knots per second: 50
	Do bias motion regularization: True
	Blake-Zisserman on reprojection errors -1
	Acceleration Huber width (sigma): -1.000000
	Gyroscope Huber width (sigma): -1.000000
	Do time calibration: False
	Max iterations: 30
	Time offset padding: 0.010000

Estimating imu-camera rotation prior
Traceback (most recent call last):
  File "/home/wzm/catkin_ws/devel/bin/kalibr_calibrate_imu_camera", line 6, in <module>
    exec(fh.read())
  File "<string>", line 236, in <module>
  File "<string>", line 197, in main
  File "/home/wzm/catkin_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py", line 109, in buildProblem
    self.CameraChain.findOrientationPriorCameraChainToImu(self.ImuList[0])
  File "/home/wzm/catkin_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 477, in findOrientationPriorCameraChainToImu
    self.camList[0].findOrientationPriorCameraToImu( imu )
  File "/home/wzm/catkin_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 134, in findOrientationPriorCameraToImu
    poseSpline = self.initPoseSplineFromCamera( timeOffsetPadding=0.0 )
  File "/home/wzm/catkin_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 288, in initPoseSplineFromCamera
    times = np.hstack((times[0] - (timeOffsetPadding * 2.0), times, times[-1] + (timeOffsetPadding * 2.0)))
IndexError: index out of bounds

ahuer2435, 2018-10-22:
你好，请问你这个问题解决了吗，同样遇到这个问题

leadorzf, 2019-10-20:
将ImageDatasetReader.py中117行img_data = np.array(self.CVB.imgmsg_to_cv(data))修改为
img_data = np.squeeze(np.array(self.CVB.imgmsg_to_cv2(data, "mono8")))

goldbattle, 2022-05-08:
See PR #15