ethz-asl/kalibr #115: Kalibr convention for pixel positions

NikolausDemmel, 2017-07-11:
What is the convention used for pixel coordinates in Kalibr? I.e., when interpreting the image as a continuous function on the image plane, does the value on pixel position (1,1) refer to the integral between (0,0) and (1,1), or something else (e.g. the integral between (0.5, 0.5) and (1.5, 1.5))?

See also the section `That strange "0.5" offset` on https://github.com/JakobEngel/dso for further background.

caselitz, 2017-07-13:
I had exactly the same question when I read this on Jakobs page. Unfortunately not sure about the answer, but here are my thoughts...

Actually the question is directly related to the assumption where the default image center is, right? Here are three options:

For integral (0.0, 0.0) to (1.0, 1.0):

1. (w/2, h/2) if index starts with 1
2. (w/2-1, h/2-1) if index starts with 0

For integral (0.5, 0.5) to (1.5, 1.5):

3. (w/2-0.5, h/2-0.5) if index starts with 0

I was also wondering how OpenCV calibration is doing it and found [this note](http://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html#calibratecamera) that mentions (w/2-0.5,h/2-0.5), i.e., the 3. option.

In Kalibr I found [this code](https://github.com/ethz-asl/kalibr/blob/master/aslam_cv/aslam_cameras/include/aslam/cameras/implementation/PinholeProjection.hpp#L716) for the intrinsic parameter initialization. It says ((w-1)/2, (h-1)/2). 
As this is = (w/2-0.5, h/2-0.5), i.e., also the 3. option, my guess is that Kalibr uses the integral (0.5, 0.5) to (1.5, 1.5).
However, it would be nice if someone could confirm (or deny) this!

[Maybe option 1. is used by some Matlab toolboxes as Matlab is 1-index based?]

NikolausDemmel, 2017-07-14:
Thanks Tim for the pointers, however it adds to the confusion that Jakob's statement is somewhat ambiguous and doesn't distinguish properly between pixel positions and image coordinates. I take his "value on pixel position (1,1)" to mean the value of the top-left pixel. The question is, what image coordinates does this value get, and by that I mean what image-plane coordinates does the center of the pixel have (in unit "pixels").

I believe there are actually 3 reasonable choices: That pixel-center gets coordinates (0,0), (0.5, 0.5), (1,1). For clarity I tabulated the corresponding image coordinates of the center of the top-left pixel, of the top-left corner of the image, of the bottom-left corner and pxiel of the image, and of the image center.

|    | Top-left pixel center | top-left pixel integral range | top-left image corner | bottom-right pixel center | bottom-right image corner | image center       |
|----|-----------------------|-------------------------------|-----------------------|---------------------------|---------------------------|--------------------|
| a) | (0,0)                 | [-0.5; 0.5]                   | (-0.5,-0.5)           | (w-1, h-1)                | (w-0.5, h-0.5)            | ((w-1)/2, (h-1)/2) |
| b) | (0.5,0.5)             | [0; 1]                        | (0,0)                 | (w-0.5, h-0.5)            | (w, h)                    | (w/2, h/2)         |
| c) | (1,1)                 | [0.5; 1.5]                    | (0.5,0.5)             | (w, h)                    | (w+0.5, h+0.5)            | ((w+1)/2, (h+1)/2) |

It seems that Jakob's choices actually correspond to b) and c), but also that the hints in OpenCV and Kalibr suggest a). This also doesn't coincide with your statement, so please point out to me if I'm going wrong somewhere...

I think we'll have to check in DSO code how this is actually interpreted. As for Kalibr or OpenCV, I guess it has to boil down to how the calibration target detectors define image coordinates, i.e. what coordinates they give if one of the corners of the calibration targets lies for example exactly in the image center (and of course if these coordinates are manipulated by Kalibr / OpenCV, or taken as is).

NikolausDemmel, 2017-07-14:
Moreover, from the Readme it seems DSO uses b), so starting from c) you need to apply the -0.5 offset that is in the code. However, starting from a), as might be the case for Kalibr / OpenCV, you need to apply +0.5 instead.

caselitz, 2017-07-14:
I've to admit that I'm bit confused - as you said, the statement is somewhat ambiguous: I actually assumed that `pixel at integer position (1,1)` on the DSO page meant not the upper-left but the 2nd pixel from the upper-left corner (i.e., the upper-left would be position (0,0)). However, only my choice 2. is valid in this case and corresponds to a). My choice 1. only make sense if position (1,1) means the upper-left pixel and in this case corresponds to b). So I think my choices where not consistent.

Thanks for the overview, I think it's pretty helpful here. I agree that these are reasonable choices. One might add

  | Top-left pixel center | top-left pixel integral range | top-left image corner | bottom-right pixel center | bottom-right image corner | image center
-- | -- | -- | -- | -- | -- | --
d) | (1.5,1.5) | [1; 2] | (1,1) | (w+0.5, h+0.5) | (w+1, h+1) | (w/2+1, h/2+1)

to do it similar to b) but having a 1-based instead of 0-based index (like Matlab).

I agree that OpenCV and Kalibr probably correspond to a) since this is the case where the image center fits to the pointers I posted.

Assuming `pixel at integer position (1,1)` means the upper-left pixel, DSO uses c) according to the description, right?

> DSO uses the convention that the pixel at integer position (1,1) in the image contains the integral over the continuous image function from **(0.5,0.5) to (1.5,1.5)**

As Jakob assumes that the given calibration uses b)

> contains the integral over (0.5,0.5) to (1.5,1.5), or the integral over **(1,1) to (0,0)**. The above conversion assumes that the given calibration in the calibration file uses the **latter convention**

he applies the -0.5 correction to go from b) to c). It seems confusing that it is **-** 0.5 but isn't this because you shift the pixel coordinate system and not the image point?

Following this logic one has to apply a -1.0 correction from a) Kalibr to c) DSO.

NikolausDemmel, 2017-07-14:
Yes, you are right, that assuming `pixel at integer position (1,1)` means the upper-left pixel, DSO uses c). I mixed it up.

Now as you say, let's assume we have a calibration in convention b) and want to change it to convention c). Let's assume that we have a perfectly centered camera, so the calibration will be

```
cx_old = w/2
cy_old = h/2
```

meaning that if we project a 3D coordinate (0,0,z) into the image plane, we will get exactly the image center (w/2, h/2). However, now we want to update this to convention c), meaning that the same point (0,0,z) should still be mapped to the image center, no having pixel coordinates ((w+1)/2, (h+1)/2). To achieve this we need to update the given calibration with

```
cx_new = w/2+0.5 = cx_old + 0.5
cy_new = h/2+0.5 = cy_old + 0.5
```

In other words we would need to apply a `+0.5` offset in the DSO code, and not as is now a `-0.5` offset and coming from a calibration with a) it would need to be `+1.0`, right?

If however `pixel at integer position (1,1)` means "second row second column pixel", then DSO uses a). However, that would mean that the convention he assumes the calibration to be in is even "0.5 pixels before a)", meaning that the top left corner pixel would correspond to the integration interval [-1,0], which would be a bit strange in my book. However, if the text in the readme is simply inconsistent and Jakob meant to say that he uses a), but assumes input in b), then the `-0.5` offset in the code at least would make sense. We'll have to check in the code though.

caselitz, 2017-07-14:
> In other words we would need to apply a +0.5 offset in the DSO code, and not as is now a -0.5 offset and coming from a calibration with a) it would need to be +1.0, right?

I think you are right. My idea

> It seems confusing that it is - 0.5 but isn't this because you shift the pixel coordinate system and not the image point?

was more a desperate try to explain the minus. ;) It should be plus.

caselitz, 2017-07-14:
I think I found the solution, see [this comment in Jakobs code](https://github.com/JakobEngel/dso/blob/master/src/util/Undistort.cpp#L797)
> // rescale and substract 0.5 offset.
// the 0.5 is because I'm assuming the calibration is given such that the pixel at (0,0)
// contains the integral over intensity over [0,0]-[1,1], whereas I assume the pixel (0,0)
 // to contain a sample of the intensity ot [0,0], which is best approximated by the integral over
// [-0.5,-0.5]-[0.5,0.5]. Thus, the shift by -0.5.

He is indeed going from b) to a) and `pixel at integer position (1,1)` means "second row second column pixel".

To be honest, I don't understand your concern

> However, that would mean that the convention he assumes the calibration to be in is even "0.5 pixels before a)", meaning that the top left corner pixel would correspond to the integration interval [-1,0], which would be a bit strange in my book.

I think he just assumes calibration given in b) and DSO code assumes a).

JakobEngel, 2017-07-14:
Indeed, from what I understand, DSO assumes a). 
Quick example just to make sure we are on the same page:
In a 1D, perfectly symmetrical image with 10 pixels (indexed 0-9), the color of the point (0,z) is the mean of pixel 4 and 5, i.e., it is projected to cx=4.5 and then linearly interpolated.

caselitz, 2017-07-17:
Thanks for your reply Jakob! I think we're on the same page, what you said corresponds to a).

As we were just discussing the convention for the pixel coordinate system, I would say how the interpolation is done is already the next step. Probably linearly makes sense, but I could also imagine nearest neighbor...

caselitz, 2017-07-17:
Btw. which calibration toolbox did you use that gives output in b) ?

Also I think Niko is right with

> As for Kalibr or OpenCV, I guess it has to boil down to how the calibration target detectors define image coordinates, i.e. what coordinates they give if one of the corners of the calibration [...]

The hints I found are really not more than hints... they suggest that both, OpenCV and Kalibr, give results in a), but I'm still not 100% convinced. I think this should still be confirmed.

NikolausDemmel, 2017-07-17:
> To be honest, I don't understand your concern

> However, that would mean that the convention he assumes the calibration to be in is even "0.5 pixels before a)", meaning that the top left corner pixel would correspond to the integration interval [-1,0], which would be a bit strange in my book.

> I think he just assumes calibration given in b) and DSO code assumes a).

Yes. I was talking about the typo in the readme text, which I correct in https://github.com/JakobEngel/dso/pull/75

> Indeed, from what I understand, DSO assumes a).
Quick example just to make sure we are on the same page:
In a 1D, perfectly symmetrical image with 10 pixels (indexed 0-9), the color of the point (0,z) is the mean of pixel 4 and 5, i.e., it is projected to cx=4.5 and then linearly interpolated.

Yes, makes sense. Thanks for confirming. As mentioned in the Readme it is also in line with the upscaling code in https://github.com/JakobEngel/dso/blob/a9915821826609ecb87afd04d916fdd8bf224fb4/src/util/globalCalib.cpp#L94-L95.

> As we were just discussing the convention for the pixel coordinate system, I would say how the interpolation is done is already the next step. Probably linearly makes sense, but I could also imagine nearest neighbor...

Sure, it is a small tradeoff between accuracy and speed, but linear interpolation is simple enough and gives you sub-pixel accuracy, I guess.

> As for Kalibr or OpenCV, I guess it has to boil down to how the calibration target detectors define image coordinates, i.e. what coordinates they give if one of the corners of the calibration [...]

> The hints I found are really not more than hints... they suggest that both, OpenCV and Kalibr, give results in a), but I'm still not 100% convinced. I think this should still be confirmed.

Indeed.

NikolausDemmel, 2017-07-17:
A quick check with an artificial image also suggests that opencv chessboard detector at least follows convention a). A corner lying exactly between pixel `(19,19)` and `(20, 20)` gets detected with coordinates `(19.5, 19.5)`.

```
import cv2
img = cv2.imread("chessboard4x4.png", 0)
print(cv2.findChessboardCorners(img, (3,3))[1][0])

==>

[[ 19.5  19.5]]
```

[image]

NikolausDemmel, 2017-07-17:
I did something similar for Kalibr, where I created an april tag, imported it in Gimp without anti aliasing, then created a bag file for Kalibr and ran the calibration with some added output of april tag corder detection coordinates, which also confirms that the corners that lie exactly between pixels have the same `0.5` image coordinates that fit with convention a).

The question that remains is if Kalibr and/or OpenCV for some reason maybe later add an offset.

@JakobEngel, do you know which calibration toolboxes actually follow b) like you assume?