ethz-asl/kalibr #116: RuntimeError: Optimization failed!

derekhe89, 2017-07-12:
when I used kalibr calibrate imu and camera  ,it appeared " RuntimeError: Optimization failed!",my imu frequency is 60hz and camera requency is 20hz.
       The message printed by the terminal is as follows：
importing libraries
Initializing IMUs:
  Update rate: 60.0
  Accelerometer:
    Noise density: 0.08 
    Noise density (discrete): 0.619677335393 
    Random walk: 0.1
  Gyroscope:
    Noise density: 0.012
    Noise density (discrete): 0.092951600309 
    Random walk: 0.03
Initializing imu rosbag dataset reader:
	Dataset:          /home/hl/kalibr_workspace/mynteye_cameraimu_calib_711_2.bag
	Topic:            /imu0
bagstart 38726.1760527
baglength 69.95136093
[ WARN] [1499821758.521938]: BagImuDatasetReader: truncated 1801 / 4168 messages.
	Number of messages: 4168
Reading IMU data (/imu0)
  Read 2367 imu readings over 40.0 seconds                       
Initializing calibration target:
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.024 [m]
    Spacing 0.0072 [m]
Initializing camera chain:
Camera chain - cam0:
  Camera model: pinhole
  Focal length: [365.9021105354135, 371.1036033722559]
  Principal point: [385.6213176029975, 231.80948369303306]
  Distortion model: radtan
  Distortion coefficients: [-0.20973302237135366, 0.050948220015473986, 0, 0]
  baseline: no data available
Camera chain - cam1:
  Camera model: pinhole
  Focal length: [365.9021105354135, 371.1036033722559]
  Principal point: [359.5612415709596, 245.46363443410945]
  Distortion model: radtan
  Distortion coefficients: [-0.20607131885372323, 0.028903158396448237, 0.00180571880675569, -0.00553860461961384]
  baseline: [[   0.99981575   -0.00194901    0.01909619 -118.64005405]
 [   0.00190259    0.99999519    0.00244903   -0.38026237]
 [  -0.01910087   -0.00241225    0.99981465    0.76019651]
 [   0.            0.            0.            1.        ]]
Initializing camera rosbag dataset reader:
	Dataset:          /home/hl/kalibr_workspace/mynteye_cameraimu_calib_711_2.bag
	Topic:            /cam0/image_raw
[ WARN] [1499821765.038307]: BagImageDatasetReader: truncated 597 / 1387 images.
	Number of images: 1387
Extracting calibration target corners
  Extracted corners for 790 images (of 790 images)                              
Initializing camera rosbag dataset reader:
	Dataset:          /home/hl/kalibr_workspace/mynteye_cameraimu_calib_711_2.bag
	Topic:            /cam1/image_raw
[ WARN] [1499821792.498072]: BagImageDatasetReader: truncated 597 / 1387 images.
	Number of images: 1387
Extracting calibration target corners
  Extracted corners for 790 images (of 790 images)                              
Baseline between cam0 and cam1 set to:
T=  [[   0.99981575   -0.00194901    0.01909619 -118.64005405]
 [   0.00190259    0.99999519    0.00244903   -0.38026237]
 [  -0.01910087   -0.00241225    0.99981465    0.76019651]
 [   0.            0.            0.            1.        ]]
Baseline:  118.64309893  [m]

Building the problem
	Spline order: 6
[... 113 lines cut ...]
[30]: J: 3.4804e+09, dJ: 4.38844e+08, deltaX: 35.1416, LM - lambda:172.332 mu:2
[ERROR] [1499821955.400941]: Optimization failed!
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_calibrate_imu_camera", line 5, in <module>
    exec(fh.read())
  File "<string>", line 206, in <module>
  File "<string>", line 182, in main
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py", line 181, in optimize
    raise RuntimeError("Optimization failed!")
RuntimeError: Optimization failed!
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils

Original exception was:
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_calibrate_imu_camera", line 5, in <module>
    exec(fh.read())
  File "<string>", line 206, in <module>
  File "<string>", line 182, in main
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py", line 181, in optimize
    raise RuntimeError("Optimization failed!")
RuntimeError: Optimization failed!

yangtzey, 2018-09-04:
I have the same problem with you. Have you solved this problem?

vishals1010, 2020-11-02:
hi, have you solved this issue ....?

lapchun, 2021-04-09:
It's mainly because you don't record the .bag file in a right way.
Maybe this will help:
https://github.com/ethz-asl/kalibr/wiki/camera-imu-calibration

yangtzey, 2022-04-22:
这是来自QQ邮箱的假期自动回复邮件。
自动回复