ethz-asl/kalibr #117: how to adjust the approx. sync. tolerance

wzhouuu, 2017-07-18:
I'm running the multiple camera calibration in ubuntu 16.04. I've tested the sample dataset, which worked fine and produced reports. So I recorded my own rosbag with three cameras. I also changed the tagSize to 0.034 in yaml file.
When I ran 
kalibr_calibrate_cameras --target src/april_6x6_80x80cm.yaml --bag multi_cam.bag --models pinhole-equi pinhole-equi pinhole-equi --topics /camera/image/A0 /camera/image/A1 /camera/image/A2
it initialized each camera and extracted corners. But after that, it gave an error "Cameras are not connected through mutual observations, please check the dataset. Maybe adjust the approx. sync. tolerance."
File "/home/wz/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/MulticamGraph.py", line 308, in plotGraph
    edge_label=self.G.es["weight"],
KeyError: 'Attribute does not exist'
I would assume this is due to the rosbag I recorded. But how should I fix this problem? 
Thanks!

wzhouuu, 2017-07-19:
Problem fixed. The timestamps for my three camera topics are quite different from each other. Solved problem by forcing the timestamps to be the same.

closed 2017-07-19 by wzhouuu

xyflzyn, 2022-03-19:
@ghost Excuse me, I met the same question. Could you please tell me how to force the timestamps to be the same time. Thank you very much!