ethz-asl/kalibr #122: no module name numpy_eigen when running kalibr_calibrate_imu_camera sample dataset

ahundt, 2017-07-28:
I tried running the example from the wiki and I'm running into trouble with numpy_eigen and libnumpy_eigen. Do you have any suggestions on how to configure it correctly?

```
kalibr_calibrate_imu_camera --target april_6x6.yaml --cam camchain.yaml --imu imu_adis16448.yaml --bag dynamic.bag --bag-from-to 5 45
importing libraries
Traceback (most recent call last):
  File "/Users/athundt/source/robone_ws/devel/bin/kalibr_calibrate_imu_camera", line 6, in <module>
    exec(fh.read())
  File "<string>", line 3, in <module>
  File "/Users/athundt/source/robone_ws/devel/lib/python2.7/site-packages/sm/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
  File "/Users/athundt/src/robone_ws/install/lib/python2.7/site-packages/numpy_eigen/__init__.py", line 5, in <module>
    from libnumpy_eigen import *
ImportError: No module named libnumpy_eigen
```

One issue is that it may be trying to open `.so` files when I've generated `.dylib` files. (I'm on OS X 10.11)

iliis, 2017-10-31:
I've had the same issue on Linux, but for me the problem was that the compiled libraries were placed in `ws/devel/lib/python2.7/dist-packages` instead of `.../site-packages`.

Some of the libraries are referenced trough hard-coded filenames, so you might try to change these to use `.dylib` instead of `.so` in:

```
Kalibr/aslam_cv/aslam_cv_python/python/aslam_cv/__init__.py
Kalibr/aslam_cv/aslam_cameras_april/python/aslam_cameras_april/__init__.py
Kalibr/aslam_incremental_calibration/incremental_calibration_python/src/incremental_calibration/__init__.py
```

fmohammadi2g, 2018-01-30:
@ahundt 
Did you solve the problem, I have the same problem. 
Thanks

closed 2022-05-03 by goldbattle