ethz-asl/kalibr #125: Problem with IMU intrinsic calibration

allenyoung25, 2017-08-04:
if I used `kalibr_calibrate_imu_camera` to calibrate IMU intrinsic with the initial config:
``

allenyoung25, 2017-08-04:
If I change the IMU-models like `kalibr_calibrate_imu_camera --target april_target.yaml --bag april.bag --imu imu.yaml --imu-models scale-misalignment --cam cam.yaml`
The results is still not changed:
> IMU0:
> ----------------------------
>  Model: scale-misalignment
  Update rate: 100.0
  Accelerometer:
    Noise density: 0.2 
    Noise density (discrete): 2.0 
    Random walk: 0.02
  Gyroscope:
    Noise density: 0.05
    Noise density (discrete): 0.5 
    Random walk: 4e-05

My `april_target.yaml`:
>   target_type: 'aprilgrid' #gridtype
  tagCols: 6                  #number of apriltags
  tagRows: 6                  #number of apriltags
  tagSize: 0.03               #size of apriltag, edge to edge [m]
  tagSpacing: 0.5             #ratio of space between tags to tagSize

`imu.yaml`:
> rostopic: /imu0
update_rate: 100.0 #Hz

>accelerometer_noise_density: 0.2 #continous
accelerometer_random_walk: 0.02 
gyroscope_noise_density: 0.05 #continous
gyroscope_random_walk: 4.0e-5

`cam.yaml`:
> cam0:
  cam_overlaps: []
  camera_model: pinhole
  distortion_coeffs: [0.238299, -0.471433, -0.002557, 0.002106]
  distortion_model: radtan
  intrinsics: [644.342698, 641.877983, 234.480426, 318.734576]
  resolution: [480, 640]
  rostopic: /cam0/image_raw

ttomchy, 2018-01-07:
Hi ,I am also meet  the problem ,did you solve it ?

shilei001, 2019-04-24:
Did you solve it？

closed 2022-04-27 by goldbattle