ethz-asl/kalibr #127: RuntimeError: TimeWentBackwardsException

voyageanp, 2017-08-08:
I have got this runtimeerror, and could you kindly teach me how to remove this error 
Best
```
RuntimeError: [TimeWentBackwardsException] /home/kohei/catkin_ws/src/kalibr_workspace/src/Kalibr/Schweizer-Messer/sm_timing/include/sm/timing/implementation/TimestampCorrector.hpp:26: correctTimestamp() assert(remoteTime > _convexHull[_convexHull.size() - 1].x) failed [1.50218e+09 > 1.50218e+09]: The correction algorithm requires that times are passed in with monotonically increasing remote timestamps
```

vishals1010, 2020-10-30:
is this issue solved, i get the same issue

goldbattle, 2022-04-22:
Correct your driver as it is publishing timestamps in a non-monotonically fashion.

closed 2022-04-22 by goldbattle