ethz-asl/kalibr #128: RuntimeError:  initSplineSparse() assert(numSegments >= 1) failed [0 >= 1]

JiayuYANG, 2017-08-10:
I got this runtime error when trying to calibrate IMU and camera:
`@ubuntu:~/FYPI$ ./kalibr-cde/kalibr_calibrate_imu_camera --bag ./rawdata/nexus6p_2_1920_2fps_3min.bag --cam camchain-homeexbotFYPIrawdatanexus6p_2_1920_2fps_3min.yaml --imu imu.yaml --target april_6x6_50x50cm.yaml 
importing libraries
Initializing IMUs:
  Update rate: 200.0
  Accelerometer:
    Noise density: 0.0012 
    Noise density (discrete): 0.0169705627485 
    Random walk: 0.000433
  Gyroscope:
    Noise density: 0.00012
    Noise density (discrete): 0.00169705627485 
    Random walk: 2.66e-05
Initializing imu rosbag dataset reader:
	Dataset:          /home/exbot/FYPI/rawdata/nexus6p_2_1920_2fps_3min.bag
	Topic:            /imu0
	Number of messages: 50445
Reading IMU data (/imu0)
  Read 50445 imu readings over 0.0 seconds                            
Initializing calibration target:
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.055 [m]
    Spacing 0.0165 [m]
Initializing camera chain:
Camera chain - cam0:
  Camera model: pinhole
  Focal length: [1464.1782284008282, 1463.6135687576389]
  Principal point: [970.2516300824147, 516.8690947315039]
  Distortion model: equidistant
  Distortion coefficients: [0.5600883361522624, -1.346201485247712, 3.0694581091836084, 1.149939528560311]
  baseline: no data available
Initializing camera rosbag dataset reader:
	Dataset:          /home/exbot/FYPI/rawdata/nexus6p_2_1920_2fps_3min.bag
	Topic:            /cam0/image_raw
	Number of images: 271
Extracting calibration target corners
  Extracted corners for 254 images (of 271 images)                              

Building the problem
	Spline order: 6
	Pose knots per second: 70
	Do pose motion regularization: False
		xddot translation variance: 1000000.000000
		xddot rotation variance: 100000.000000
	Bias knots per second: 50
	Do bias motion regularization: True
	Blake-Zisserman on reprojection errors -1
	Acceleration Huber width (m/s^2): -1.000000
	Gyroscope Huber width (rad/s): -1.000000
	Do time calibration: False
	Max iterations: 30
	Time offset padding: 0.020000

Estimating imu-camera rotation prior

Initializing a pose spline with 0 knots (100.000000 knots per second over 0.000177 seconds)
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_calibrate_imu_camera", line 5, in <module>
    exec(fh.read())
  File "<string>", line 206, in <module>
  File "<string>", line 172, in main
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py", line 113, in buildProblem
    self.CameraChain.findOrientationPriorCameraChainToImu(self.ImuList[0])
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 438, in findOrientationPriorCameraChainToImu
    self.camList[0].findOrientationPriorCameraToImu( imu )
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 122, in findOrientationPriorCameraToImu
    poseSpline = self.initPoseSplineFromCamera( timeOffsetPadding=0.0 )
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 276, in initPoseSplineFromCamera
    pose.initPoseSplineSparse(times, curve, knots, 1e-4)
RuntimeError: [Exception] /var/kalibr-build/src/kalibr/aslam_nonparametric_estimation/bsplines/src/BSpline.cpp:970: initSplineSparse() assert(numSegments >= 1) failed [0 >= 1]: There must be at least one time segment
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils
`

Could anybody tell me what's going on and how to fix it?

DreamCatcher096, 2017-09-01:
Are you sure your timestamp is measured in nanoseconds?

WR0903, 2019-01-12:
wr@wr:~/kalibr_workspace/src/q$ kalibr_calibrate_imu_camera --target april_6x6_80x80cm.yaml --cam camchain.yaml --imu imu_adis16448.yaml --bag 2019-01-12-06-17-31.bag --bag-from-to 0 30
importing libraries
Initializing IMUs:
  Update rate: 70
  Accelerometer:
    Noise density: 0.01 
    Noise density (discrete): 0.0836660026534 
    Random walk: 0.0002
  Gyroscope:
    Noise density: 0.005
    Noise density (discrete): 0.0418330013267 
    Random walk: 4e-06
Initializing imu rosbag dataset reader:
	Dataset:          2019-01-12-06-17-31.bag
	Topic:            /imu/data
bagstart 1547273851.35
baglength 87.6256062984
[ WARN] [1547304655.818156]: BagImuDatasetReader: truncated 4109 / 6252 messages.
	Number of messages: 6252
Reading IMU data (/imu/data)
  Read 2143 imu readings over 30.0 seconds                   
Initializing calibration target:
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.03 [m]
    Spacing 0.009 [m]
Initializing camera chain:
Camera chain - cam0:
  Camera model: pinhole
  Focal length: [314.6187799011879, 320.2956368270001]
  Principal point: [379.9776011718409, 225.8942986221469]
  Distortion model: equidistant
  Distortion coefficients: [0, 0, 0, 0]
  baseline: no data available
Initializing camera rosbag dataset reader:
	Dataset:          2019-01-12-06-17-31.bag
	Topic:            /frame
[ WARN] [1547304656.642365]: Bag end time of 30.0 s is bigger than the total length of 0.0 s
[ WARN] [1547304656.647354]: BagImageDatasetReader: truncated 0 / 2629 images.
	Number of images: 2629
Extracting calibration target corners
  Extracted corners for 2068 images (of 2629 images)                              

Building the problem
	Spline order: 6
	Pose knots per second: 100
	Do pose motion regularization: False
		xddot translation variance: 1000000.000000
		xddot rotation variance: 100000.000000
	Bias knots per second: 50
	Do bias motion regularization: True
	Blake-Zisserman on reprojection errors -1
	Acceleration Huber width (sigma): -1.000000
	Gyroscope Huber width (sigma): -1.000000
	Do time calibration: True
	Max iterations: 30
	Time offset padding: 0.030000
Estimating time shift camera to imu:

Initializing a pose spline with 0 knots (100.000000 knots per second over 0.000000 seconds)
Traceback (most recent call last):
  File "/home/wr/kalibr_workspace/devel/bin/kalibr_calibrate_imu_camera", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/wr/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 236, in <module>
    main()
  File "/home/wr/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 197, in main
    verbose = parsed.verbose)
  File "/home/wr/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py", line 105, in buildProblem
    cam.findTimeshiftCameraImuPrior(self.ImuList[0], verbose)
  File "/home/wr/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 219, in findTimeshiftCameraImuPrior
    poseSpline = self.initPoseSplineFromCamera( timeOffsetPadding=0.0 )
  File "/home/wr/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 313, in initPoseSplineFromCamera
    pose.initPoseSplineSparse(times, curve, knots, 1e-4)
RuntimeError: [Exception] /home/wr/kalibr_workspace/src/Kalibr/aslam_nonparametric_estimation/bsplines/src/BSpline.cpp:970: initSplineSparse() assert(numSegments >= 1) failed [0 >= 1]: There must be at least one time segment

epiception, 2019-02-06:
Hey, any updates on this?

vishals1010, 2020-11-02:
hi, have you solved this issue ....?

goldbattle, 2022-04-22:
It looks like your timestamps are messed up, see
```
Read 50445 imu readings over 0.0 seconds
```
and
```
Initializing a pose spline with 0 knots (100.000000 knots per second over 0.000000 seconds)
```

I recommend you take a look at the raw data, and go from there.
Both camera and IMU need to be using the same reference clock frame (there can be an offset of course).

closed 2022-05-19 by goldbattle