ethz-asl/kalibr #136: ZR300 optimization failed

Cezary-Ziemiecki, 2017-10-17:
Hi,
I am trying to calibrate camera RealSense ZR300 and I've got this runtime error:

```
 importing libraries
Initializing IMUs:
  Update rate: 200.0
  Accelerometer:
    Noise density: 0.01 
    Noise density (discrete): 0.141421356237 
    Random walk: 0.0002
  Gyroscope:
    Noise density: 0.005
    Noise density (discrete): 0.0707106781187 
    Random walk: 4e-06
Initializing imu rosbag dataset reader:
	Dataset:          bag4.bag
	Topic:            /camera/imu/data_raw
	Number of messages: 16057
Reading IMU data (/camera/imu/data_raw)
  Read 16057 imu readings over 41.2 seconds                          
Initializing calibration target:
  Type: aprilgrid
  Tags: 
    Rows: 3
    Cols: 3
    Size: 0.047 [m]
    Spacing 0.0141 [m]
Initializing camera chain:
Camera chain - cam0:
  Camera model: pinhole
  Focal length: [306.179, 306.929]
  Principal point: [158.523, 122.748]
  Distortion model: radtan
  Distortion coefficients: [0.06696815, -0.11570779, 0.00017277, -0.00054532]
  baseline: no data available
Initializing camera rosbag dataset reader:
	Dataset:          bag4.bag
	Topic:            /camera/color/image_raw
	Number of images: 619
Extracting calibration target corners
  Extracted corners for 502 images (of 619 images)                              

Building the problem
	Spline order: 6
	Pose knots per second: 100
	Do pose motion regularization: False
		xddot translation variance: 1000000.000000
		xddot rotation variance: 100000.000000
	Bias knots per second: 50
	Do bias motion regularization: True
	Blake-Zisserman on reprojection errors -1
	Acceleration Huber width (sigma): -1.000000
	Gyroscope Huber width (sigma): -1.000000
	Do time calibration: False
	Max iterations: 60
	Time offset padding: 0.010000

Estimating imu-camera rotation prior

Initializing a pose spline with 4120 knots (100.000000 knots per second over 41.200000 seconds)
Gravity was intialized to [-3.35176344 -7.36549785 -5.53927307] [m/s^2]
  Orientation prior camera-imu found as: (T_i_c)
[[ 0.99910519 -0.0262021  -0.03320044]
 [ 0.02384797  0.9973024  -0.06942038]
 [ 0.03492984  0.0685665   0.99703488]]
  Gyro bias prior found as: (b_gyro)
[-0.18099904  0.17929133 -0.07491421]

Initializing a pose spline with 4124 knots (100.000000 knots per second over 41.240000 seconds)
[... 107 lines cut ...]
[52]: J: 5.43182e+14, dJ: 8.85602e+09, deltaX: 1.17357e-05, LM - lambda:4.9286e+09 mu:2
[53]: J: 5.43173e+14, dJ: 9.37377e+09, deltaX: 1.26221e-05, LM - lambda:7.72498e+09 mu:2
[54]: J: 5.43145e+14, dJ: 2.75419e+10, deltaX: 1.08178e-05, LM - lambda:1.21233e+10 mu:2
[55]: J: 5.43129e+14, dJ: 1.68309e+10, deltaX: 7.64468e-06, LM - lambda:1.28254e+10 mu:2
[56]: J: 5.43119e+14, dJ: 9.35821e+09, deltaX: 5.12794e-06, LM - lambda:1.30747e+10 mu:2
[57]: J: 5.43114e+14, dJ: 5.46634e+09, deltaX: 3.42053e-06, LM - lambda:1.31112e+10 mu:2
[58]: J: 5.4311e+14, dJ: 3.75373e+09, deltaX: 2.24482e-06, LM - lambda:1.31072e+10 mu:2
[59]: J: 5.43107e+14, dJ: 3.39583e+09, deltaX: 1.67594e-06, LM - lambda:1.19027e+10 mu:2
[60]: J: 5.43081e+14, dJ: 2.57645e+10, deltaX: 1.47878e-05, LM - lambda:3.96755e+09 mu:2
[ERROR] [1508215661.949494]: Optimization failed!
Traceback (most recent call last):
  File "/root/kalibr_workspace/devel/bin/kalibr_calibrate_imu_camera", line 6, in <module>
    exec(fh.read())
  File "<string>", line 236, in <module>
  File "<string>", line 206, in main
  File "/root/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py", line 179, in optimize
    raise RuntimeError("Optimization failed!")
RuntimeError: Optimization failed!

```
For sure, I set max_iter to high value (60000) and left my office. Today's morning it was still calibrate 
[image] .
Do you have an idea what could cause a calibration problem?
It could be wrong bag, Apriltag target or cam/imu_config.
By the way, Did someone calibrate this camera and have the appropriate data to create a config file? I think that bad transform data between imu and camera is a main reason.

CtfChan, 2018-03-10:
Hey I'm having the same problem. Did you figure out where your issue was?

Mcthomas777, 2021-04-29:
Hey, I'm facing these issues too...Did you guys found something out there, I checked my IMU samples (good units, good looking), pictures too... so I don't know what to do

z1047941150, 2022-01-08:
Please help us. So many wrongs.     :(