ethz-asl/kalibr #151: Questions about the IMU config files.

ttomchy, 2018-01-07:
Hello ,I am using the IMU MTI-30-2A5G4 , in the datasheet https://www.xsens.com//download/pdf/documentation/mti-10/mti-10-series.pdf , in the datasheet, I can only find the Noise density about Gyroscopes Accelerometers. But I do not know how to get the accelerometer_random_walk and the gyroscope_random_walk ? Is anyone know how to get those two parameters?

cynosure4sure, 2018-03-31:
You can record a data set for your IMU and use a tool to plot alan deviation such as https://www.mathworks.com/matlabcentral/fileexchange/55765-avar
to get your parameters according to the IMU Noise model in the wiki of this git.

goldbattle, 2022-05-03:
See the wiki for details:
https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model

closed 2022-05-03 by goldbattle