ethz-asl/kalibr #153: problem with undistortion by OpenCV

imrasp, 2018-01-17:
I have tried to undistort the image from[ Euroc dataset](http://robotics.ethz.ch/~asl-datasets/ijrr_euroc_mav_dataset/calibration_datasets/) using intrinsic and distortion coefficient provided by the dataset itself which using Kalibr as a calibration tool. The result is completely distorted as follow:

original image:
[image]
undistorted image:
[image]

The method for unsidtort is (provided by opencv)
```
cv::undistort(image, imageUndistorted, intrinsic, distCoeffs);
```
I have try to undistort the image capture on my action camera and calibrated by [opencv camera calibration](https://github.com/opencv/opencv/blob/2.4/samples/cpp/calibration.cpp). the result as follow:

original image:
[image]
undistorted image:
[image]

I tried recalibrate Euroc using calibration dataset on kalibr and got the same result as it provided in the datasets.

**What did I do wrong on undistort the image and what is the different between Kalibr and Opencv for calibrating monocular camera**

schneith, 2018-01-18:
Maybe the models between Kalibr/OpenCV are not matching. Kalibr implements distortion models that are not supported by OpenCV (e.g. the 'equidistant' model). Could you verify that the models are correct?

imrasp, 2018-01-18:
@schneith Please suggest me a bit more on this, I am new to these calibration methods. Kalibr seems to support all distortion model [(this link)](https://github.com/ethz-asl/kalibr/wiki/supported-models)  including radial tangential which is the same as [opencv distortion model](https://docs.opencv.org/2.4/doc/tutorials/calib3d/camera_calibration/camera_calibration.html). This is some context I copied from opencv website:
```
For the distortion OpenCV takes into account the radial and tangential factors.
```
This is the camera specification proposed by Euroc dataset
```
rate_hz: 20
resolution: [752, 480]
camera_model: pinhole
intrinsics: [458.654, 457.296, 367.215, 248.375] #fu, fv, cu, cv
distortion_model: radial-tangential
distortion_coefficients: [-0.28340811, 0.07395907, 0.00019359, 1.76187114e-05]
```
I think Euroc using the same distortion model as opencv but using these value I got the result above.

HTLife, 2018-05-30:
Is there any way to convert kalibr distortion model to OpenCV supported format?

HTLife, 2018-05-30:
@NikolausDemmel  Thank you!  I can now calibrate image with kalibr format yaml for by OpenCV.

The sample project is provided here for someone who might need it in the future.
https://github.com/HTLife/tumvi_fisheye_calib