ethz-asl/kalibr #155: Problem with long integer overflow

tuandle, 2018-01-21:
Hi,

I tried to perform camera-imu calibration. I recorded a couple of bags. However, I could only use 1 bag for camera-imu calibration. Running kalibr_calibrate_imu_camera with other bags always returned the following error:
`Initializing a pose spline with 151645399431 knots (100.000000 knots per second over 1516453994.310450 seconds)
Traceback (most recent call last):
  File "/home/tuan/kalibr_ws/devel/bin/kalibr_calibrate_imu_camera", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/tuan/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 236, in <module>
    main()
  File "/home/tuan/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 197, in main
    verbose = parsed.verbose)
  File "/home/tuan/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py", line 105, in buildProblem
    cam.findTimeshiftCameraImuPrior(self.ImuList[0], verbose)
  File "/home/tuan/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 219, in findTimeshiftCameraImuPrior
    poseSpline = self.initPoseSplineFromCamera( timeOffsetPadding=0.0 )
  File "/home/tuan/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 313, in initPoseSplineFromCamera
    pose.initPoseSplineSparse(times, curve, knots, 1e-4)
**OverflowError: bad numeric conversion: positive overflow**`

As far as I understand, this problem relates to long integer overflow issue. Any ideas how I could fix it? 
In the meantime, I could try to record different bags to use but I am curious why this error happens to all other bag files except one.  
Thank you for your help!

Quitino, 2020-05-29:
Hi！Is this problem solved? I have the same problem

pirate999, 2020-09-22:
Hi！Is this problem solved? I have the same problem

floriantschopp, 2020-09-23:
Hi
I guess the problem here is the bag itself (and most likely the VI sensor driver) and not Kalibr. For some reason, the duration of the bag is `1516453994.310450 s` which is more than 48 years. I assume that the timestamps in your bag are messed up (e.g. IMU messages starting at Unix time 0 (which is 01/01/1970) while camera images are in the correct local time, or the other way around. This would make up for 48 years. Can you confirm this in your bags?

TouchDeeper, 2020-12-14:
@Quitino @pirate999 
Hi, 
Do you solve this problem, I have the same problem.

goldbattle, 2022-04-22:
Hi, can you provide a bag where this is a problem?

closed 2022-06-08 by goldbattle