ethz-asl/kalibr #156: Problems with IMU parameters setting

KevinGHub, 2018-01-23:
Hey guys, I am using rovio with monocular and IMU MTi 100. When I try to set my IMU parameters according to [https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model](url), I found the gyroscope_noise_density and accelerometer_noise_density here [https://www.xsens.com/download/pdf/documentation/mti-100/mti-100-series.pdf](url). As for gyroscope_random_walk, I refer to one of the papers below [http://www.acsu.buffalo.edu/~johnc/gpsins_gnc05.pdf](url) where I can't find how accelerometer_random_walk is computed. Or can  the AD graph in [https://www.xsens.com/tags/gyroscopes/](url) help in order to get accelerometer_random_walk? I have been stuck here quite a long time. Appreciate it if you can help.

KevinGHub, 2018-01-26:
I guess I have figured it out. Those're both illustrated in the wiki. So close.

closed 2018-01-26 by KevinGHub

NikolausDemmel, 2018-01-26:
Would you mind sharing your conclusion?

shilei001, 2019-04-24:
Would you mind sharing your conclusion？