ethz-asl/kalibr #158: Cannot use the cde tool  to create target pdf

ZongzhangLin, 2018-01-30:
I run kalibr-cde in both ubuntu 14.04 and 16.04, also reinstall the latex in each computer, but I still get "RuntimeError: TeX/LaTeX finished unexpectedly" when I run kalibr_create_target_pdf. 
Tried to find the answer online, but no much helpful information. Did I miss some step to setup the environment?

Thank you.

The error print out is much:
./cde-exec kalibr_create_target_pdf --type apriltag --nx 6 --ny 6 --tsize 0.08 --tspace 0.3
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/dist-packages/pyx/text.py", line 670, in run
    raise RuntimeError("TeX/LaTeX finished unexpectedly")
RuntimeError: TeX/LaTeX finished unexpectedly

still waiting for tex after 5 (of 60) seconds...
still waiting for tex after 10 (of 60) seconds...
still waiting for tex after 15 (of 60) seconds...
still waiting for tex after 20 (of 60) seconds...
still waiting for tex after 25 (of 60) seconds...
still waiting for tex after 30 (of 60) seconds...
still waiting for tex after 35 (of 60) seconds...
still waiting for tex after 40 (of 60) seconds...
still waiting for tex after 45 (of 60) seconds...
still waiting for tex after 50 (of 60) seconds...
still waiting for tex after 55 (of 60) seconds...
the timeout of 60 seconds expired and tex did not respond.
We try to finish the dvi due to an unhandled tex error
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_create_target_pdf", line 5, in <module>
    exec(fh.read())
  File "<string>", line 198, in <module>
  File "<string>", line 118, in generateAprilBoard
  File "/usr/lib/python2.7/dist-packages/pyx/canvas.py", line 268, in text
    return self.insert(self.texrunner.text(x, y, atext, *args, **kwargs))
  File "/usr/lib/python2.7/dist-packages/pyx/text.py", line 1199, in text
    self.finishdvi(ignoretail=1)
  File "/usr/lib/python2.7/dist-packages/pyx/text.py", line 1043, in finishdvi
    self.execute(None, self.defaulttexmessagesend + self.texmessagesend)
  File "/usr/lib/python2.7/dist-packages/pyx/text.py", line 1006, in execute
    self.texinput.write(self.expr)
IOError: [Errno 32] Broken pipe
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils

Original exception was:
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_create_target_pdf", line 5, in <module>
    exec(fh.read())
  File "<string>", line 198, in <module>
  File "<string>", line 118, in generateAprilBoard
  File "/usr/lib/python2.7/dist-packages/pyx/canvas.py", line 268, in text
    return self.insert(self.texrunner.text(x, y, atext, *args, **kwargs))
  File "/usr/lib/python2.7/dist-packages/pyx/text.py", line 1199, in text
    self.finishdvi(ignoretail=1)
  File "/usr/lib/python2.7/dist-packages/pyx/text.py", line 1043, in finishdvi
    self.execute(None, self.defaulttexmessagesend + self.texmessagesend)
  File "/usr/lib/python2.7/dist-packages/pyx/text.py", line 1006, in execute
    self.texinput.write(self.expr)
IOError: [Errno 32] Broken pipe
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/dist-packages/pyx/text.py", line 740, in _cleantmp
    texrunner.expectqueue.put_nowait(None)              # do not expect any output anymore
  File "/usr/lib/python2.7/Queue.py", line 148, in put_nowait
    return self.put(item, False)
  File "/usr/lib/python2.7/Queue.py", line 123, in put
    raise Full
Full
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/dist-packages/pyx/text.py", line 740, in _cleantmp
    texrunner.expectqueue.put_nowait(None)              # do not expect any output anymore
  File "/usr/lib/python2.7/Queue.py", line 148, in put_nowait
    return self.put(item, False)
  File "/usr/lib/python2.7/Queue.py", line 123, in put
    raise Full
Queue.Full
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils

Original exception was:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/dist-packages/pyx/text.py", line 740, in _cleantmp
    texrunner.expectqueue.put_nowait(None)              # do not expect any output anymore
  File "/usr/lib/python2.7/Queue.py", line 148, in put_nowait
    return self.put(item, False)
  File "/usr/lib/python2.7/Queue.py", line 123, in put
    raise Full
Queue.Full

solosuper, 2018-03-09:
@ZongzhangLin Did you resolve the problem? I have the same issue.

ZongzhangLin, 2018-04-03:
@solosuper  No, I just used the pdf they give at the end.

closed 2018-04-03 by ZongzhangLin

phil333, 2020-12-07:
I also had the 'RuntimeError: TeX/LaTeX finished unexpectedly' error. If anyone finds this bug in the future, my solution was to use `sudo apt install python-pyx` instead of `pip install pyx`