ethz-asl/kalibr #166: build error

zxp771, 2018-02-23:
Failed     << aslam_cv_backend_python:make                  [ Exited with code 2 ]                                          
Failed    <<< aslam_cv_backend_python                       [ 1 minute and 28.3 seconds ]                                   
Abandoned <<< kalibr                                        [ Unrelated job failed ]                                        
[build] Summary: 35 of 37 packages succeeded.                                                                               
[build]   Ignored:   None.                                                                                                  
[build]   Warnings:  None.                                                                                                  
[build]   Abandoned: 1 packages were abandoned.                                                                             
[build]   Failed:    1 packages failed.                                                                                     
[build] Runtime: 1 minute and 37.4 seconds total.                                           

The aslam_cv_backend_python is the only one that can not succeed.I dont know why?
Did anyone see this problem ever?

schneith, 2018-02-26:
What systems are you using? The weekly build test seems to succeed on ubuntu-trusty and ubuntu-xenial.

chiusinying, 2021-07-20:
Have you solved this problem? I can't find the solution.
Could anybody help me please?

chiusinying, 2021-07-20:
Have you solved this problem? I can't find the solution.
Could anybody help me, please?

SizheWei, 2022-03-04:
After fixing the problem of `aslam_cv_python: make` by #https://github.com/ethz-asl/kalibr/issues/276#issuecomment-486894620 
I also met this problem. I fix this problem by building again. 
I guess the reason is that the process of make is a recursive invocation. So you may need to make it more than once? 
Hope it may help you.

closed 2022-05-03 by goldbattle