ethz-asl/kalibr #169: Error when running kalibr_calibrate_rs_cameras

TheBigEyedChink, 2018-03-08:
Exception in thread block: Time is out of dual 32-bit range
I can't figure it out. Does anyone meet this problem before?

TheBigEyedChink, 2018-03-08:
This is the error in detail,
importing libraries
	Dataset:          ./rs/dataset.bag
	Topic:            /cam0/image_raw
	Number of images: 750
Extracting calibration target corners
  Extracted corners for 750 images (of 750 images)                              

Initializing a pose spline with 32 knots (100.000000 knots per second over 9.490000 seconds)
run new optimisation with initial values:

LineDelay:
9.25925925926e-05
Intrinsics:
[ 1530.17699493  1530.17699493   959.5          539.5       ]
Distortion:
[ 0.  0.  0.  0.]
Using the block_cholesky linear system solver
Using the dog_leg trust region policy
Using the block_cholesky linear system solver
Using the dog_leg trust region policy
Initializing
Optimization problem initialized with 38 design variables and 11251 error terms
The Jacobian matrix is 22501 x 219
[0.0]: J: 3.60336e+13
[1]: J: -nan, dJ: -nan, deltaX: 749.44, DL - delta:1243.13, DL, beta:0.5
run new optimisation with initial values:

LineDelay:
9.16081831586e-05
Intrinsics:
[  809.69814558   780.73660911  1101.90088177   617.95415509]
Distortion:
[ 0.00000001 -0.         -0.00000106 -0.00000029]
Using the block_cholesky linear system solver
Using the dog_leg trust region policy
Using the block_cholesky linear system solver
Using the dog_leg trust region policy
Initializing
Optimization problem initialized with 62 design variables and 11251 error terms
The Jacobian matrix is 22501 x 363
[0.0]: J: -nan
Exception in thread block: Time is out of dual 32-bit range
Exception in thread block: Time is out of dual 32-bit range
Exception in thread block: Time is out of dual 32-bit range
Exception in thread block: Time is out of dual 32-bit range
[ERROR] [1520489801.167155]: Exception: std::exception

goldbattle, 2022-04-22:
This should be fixed on master, please reopen if it is still an issue when you build from source.

closed 2022-04-22 by goldbattle

aditya7robo, 2024-06-07:
I am getting the exact same issue now in ROS Noetic, with RealSense D415 camera. 

I built the Kalibr package from source after downloading the default branch.

How do I solve it?

LeiyuanMa, 2024-12-17:
I am getting the exact same issue now when I build from dockerfile: kalibr/Dockerfile_ros1_18_04 in master branch.

Can we reopen this issue?