ethz-asl/kalibr #176: Opinion on Allan plot, derived IMU parameters and calibration result for low-cost IMU

NikolausDemmel, 2018-04-08:
We are calibrating a sensor setup with a low-cost IMU and I wanted to ask opinions on our interpretation of the results. Maybe experienced users have some insights. Also agreement or general comments would be much appreciated. I present our setup and results in the following.

We have a sensor setup with a BMI160 IMU (integrated in an Arduino) and two cameras. IMU readings are taken at 200Hz and camera images at 20. Hardware timestamps are recorded on the Arduino and a constant time-offset between cameras and IMU is calibrated offline, so the data should be time synchronized quite well (additional time-calibration during camera-imu calibration gives results in the +-0.1ms range, depending on the sequence).

We have our own IMU calibration that estimates constant scale, axis alginment, and bias offset once a priori with a model similar to the one in Kalibr. The raw IMU data is preprocessed taking into account the result from this calibration (except for the IMU sequences for the Allan plots).

In order to estimate the IMU noise parameters, we recorded a 111hour sequence of the static sensor rig, which results in the following Allan plot with our line-fits and deduced sigmas indicated.

[image]

From this we derive the following noise sigmas:

```
accelerometer white noise: 0.0013410894915444793 m/s^1.5
accelerometer bias: 8.617934702248024e-05 m/s^2.5
gyroscope white noise: 8.044388472956675e-05 rad/s^0.5
gyroscope bias: 2.1877325952252875e-06 rad/s^1.5
```

The white noise sigmas are very consistent across different static imu sequences and the -1/2 line fits well. For the biases its a bit unclearer. The 1/2 line fit is not that clear, as the dominant slope on the right hand side of the plot is a bit larger than 1/2. A reason why the bias random walk line does not fit so well and is also less consistent across different recordings might also be unmodelled effects, e.g. temperature dependence (the 111 hours were recorded in an office environment and have natural day-night cycle temperature fluctuations). This also fits the discussion in the [wiki](https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model#kalibr-imu-noise-parameters-in-practice).

For this reason we believe that the noise values from the allan plot should be inflated to give good results, the question is how much. Eventually, these noise sigmas should also be used for VI odometry/SLAM algorithms. We are aware that often you have to tune noise parameters for each algorithm anyway, but we feel like having a reasonable starting point is important.

To asses the appropriateness of noise sigmas, we are looking at the reports from Kalibr's camera-imu calibration. A result ([report-imucam-imucalib.pdf](https://github.com/ethz-asl/kalibr/files/1887427/report-imucam-imucalib.pdf)) with the original noise sigmas as above shows that calibration overall seems to work ok. Looking at the acc/gyro errors and their sigma bounds, it seems like the white noise is modelled well, however due to some occational outliers and unmodelled effects resulting in systemmatic erros (in particular noticable for the acc errors), increasing the white noise sigmas might be appropriate. Looking at the bias evolution, they are mostly in their bounds.

For the reasons above, we ran calibration with adapted sigmas. We double the white noise sigmas and we increase the random walk bias sigmas by a factor of 10, which gets us the following result: [report-imucam-imucalib.pdf](https://github.com/ethz-asl/kalibr/files/1887438/report-imucam-imucalib.pdf). It seems that now the errors are well in the sigma bounds (except for some single outliers), and the bias evolution also looks appropriate to us. The spread of the reprojection errors is slightly smaller. The result for the camera-imu intrinsics is very similar (up to around 1 mm for translation).

Would you agree that in particular the increased (x2 white noise and x10 bias random walk) noise sigmas seem to be appropriate from the calibration results, or would you suggest something else?

---

**Note on IMU calibration:** In order to make sure our IMU scale and axis-alignment calibration is appropriate, we ran the Kalibr calibration on the raw data once with Kalibr's `calibrated` IMU model (as above) and once with `scale-misalignment` (and also time-calibration), both with the inflated noise sigmas. The former shows issues in the spline fitting in particular for larger accelerations. This indicates to us that both our calibration and the one in Kalibr works fine, and that it is necessary to calibrate for this for our IMU. In fact, running Kalibr's IMU calibration again on our already pre-processed data show only slight changes suggested by Kalibr. E.g. there seems to be some disagreement on scale, but is only around 0.3%, whereas the original scale errors are calibrated to around 1-10%.

Results:
* raw data, calibrated IMU model: [report-imucam-imucalib.pdf](https://github.com/ethz-asl/kalibr/files/1887455/report-imucam-imucalib.pdf)
* raw data, scale-misalignment IMU model, time calibration: [report-imucam-imucalib.pdf](https://github.com/ethz-asl/kalibr/files/1887454/report-imucam-imucalib.pdf)
* pre-calibrated data, scale-misalignment IMU model, time calibration: [report-imucam-imucalib.pdf](https://github.com/ethz-asl/kalibr/files/1887452/report-imucam-imucalib.pdf)

zx2624, 2019-01-17:
Hi Nikolaus, from your report I've noticed that you directly applied your noise and bias data in your calibration, but I think in the yaml file the noise density should be in continuous-time, whereas the random walk does't have such requiement. Don't know if im missing something here,.

NikolausDemmel, 2019-01-29:
@zx2624, thanks for your comment. Sorry, what conversion do you suggest we are missing?